prusa3d/Prusa-Firmware #4508: New FW 3.0.0 and 3.0.1 is not compatible with MMU2S

DavidB1975, 2023-11-20:
MK3S+

MMU2S - works on FW 1.0.6. Version 3.0.0 and 3.0.1 I get MCU error

**SD card or USB/Octoprint**
  SD card

**Describe the bug**
  When i load 3.0.0 or 3.0.1 onto my MMU2S I have the first 3 green LED, 4 flashing slowly red, 5 flashing fast green. Printer after enabling MMU in the printer, display says MMU MCU Error
  
**To Reproduce**
  update MMU2S to version 3.0.0 or 3.0.1
  
**Expected behavior**
  Normal functionality after the update
  
**G-code**
None

**Video**
None

Strandkrabbe, 2023-11-25:
I have the same problem. I updated the printer firmware to 3.13.2 and the performed the Temperature Model calibration. After that I updated the MMU2S firmware to version 3.0.1. After the MMU2S resets all 5 MMU LEDs flash green irregularly. Then the idler motor starts moving really fast compared to the old firmware. After a moment it turns in the other direction and crashes into the limitation of the idler. The idler motor tries to move further and the MMU2S shuts down.

wabbitguy, 2023-11-28:
I have the MMU2S and it's running 3.0.0 just fine BUT I have to use the MK3S 3.13.0 firmware on the printer. If I update either the firmware on the printer to 3.13.2 or 3.0.1 it completely ignores the MMU2S regardless on the MMU enabled in the LCD menu or not.

DavidB1975, 2023-11-30:
Installing new FW på printer and MMU does not change anything for me, same error - I call that a bug. Works fine with 1.0.6 and 3.12.2
3.13.x and 3.0.x combo puts my MMU2S in an error state

Djamuka, 2023-12-07:
Try this :
https://help.prusa3d.com/article/factory-reset-mmu_510075

payne8, 2023-12-10:
I can confirm that the factory reset suggested by @Djamuka made my mmu2s start working. It still makes a loud noise during the self test and my mk3s prints much louder now on the 3.13.2 firmware. But it seems to be working correctly.

Djamuka, 2023-12-13:
> I can confirm that the factory reset suggested by @Djamuka made my mmu2s start working. It still makes a loud noise during the self test and my mk3s prints much louder now on the 3.13.2 firmware. But it seems to be working correctly.

FW is very agressive with MMU2s , be aware of the idler screw....

3d-gussner, 2023-12-21:
@DavidB1975 Is this issue solved if so please close it.

3d-gussner, 2023-12-28:
Closing issue due to no response.

closed 2023-12-28 by 3d-gussner

DRECKINTERNET, 2025-03-12:
Im having this same issue on printer mk3s and mmu2s with fw 3.14.1 and 3.0.3

DavidB1975, 2025-03-12:
> Im having this same issue on printer mk3s and mmu2s with fw 3.14.1 and 3.0.3

So what solved it for me was to wiggle the power connector going from the main board to the MMU. if it measures a few milivolt too little, this error occurs. I have to wiggle it from time to time, to make it work again, super annoying...... I read some has added a separate power supply to avoid the issue.

DRECKINTERNET, 2025-03-13:
So it is a power problem?

DavidB1975, 2025-03-13:
As I’v understood it is the firmware's limit that has been raised so it is
more sensitive to fluctuations, not a problem with your powesupply. A few
milivolt is enough to trigger the fault, so by cleaning the contacts
(wiggle) you regain those missing milivolts.

-- kind regards, David