prusa3d/Prusa-Firmware #4509: [FEATURE REQUEST] MK3S Option to store current settings

4rdful, 2023-11-20:
Would you implement an option to automated store all current settings?
That would be useful as

1) you want just to backup it,
2) you want to quick send Prusa's support the data for troubleshooting via eMail or
3) you want to backup the settings before doing a hard reset or a firmware update.

Here some ideas from me, but take into account that I have absolutely no idea about the code and its limitations. So just the ideas ;)

*option to store data to medium by clicking the wheel
*option to choose storage medium, like SD/ TF card or via USB connection
*USB storage to a computer could also be solved involving the PrusaSlicer if the printer itself is not capable to do it (please make sure to inform PrusaSlicer's dev team if needy)
*file's format could be ".txt" or ".json"
*format of stored information inside the file should be readable by men (not just binaries or similar types)
*a very fancy idea would be to make a restore option included, too: so first backup the data and then restore it completely to the printer (but the basic point still stays to store the settings)

Thank you for considering my thoughts!
Nice day and happy coding :)

cozmo14047, 2023-11-25:
I'd like it if it could put it onto an SD card or retrieve via gcode

3d-gussner, 2023-11-25:
Thanks for the feature request. There are few Gcode commands you can use to "backup" things:
- `M503` will report printer settings
  - that collection of gcodes can be used to restore settings 
- `M850 S0` will report your sheet settings [S1-S7]  will report the other sheets if set
  - these values can be restored using same gcode 
- `M310` your Thermal model settings (included in `M503`)
  - these values can be restored using  same gcode
- `M78` (coming in FW 3.14.0)  will report Total filament used and Total time
  - There is no "restore" or set option for the statistics and also not planed at this moment

All EEPROM values can be read and written by using `D3` gcode commands. BUT as there are some limitations like case sensitivity.
**Be very very careful using these. It may overwrite your EEPROM and a factory reset is needed!!!**
See https://prusa3d.github.io/Prusa-Firmware-Doc/Prusa-Firmware/group__eeprom__table.html 

Implementing a backup and restore function in the firmware would use lot of precious resources which we need for other features. I don't think that we will add this to the firmware.

Maybe someone from the community is interested in this feature request and provides a script or plugin or ??? to backup and restore the settings.

cozmo14047, 2023-11-25:
@3d-gussner thanks, it's a shame because the one thing alot of people want to keep are statistics.

3d-gussner, 2023-11-28:
@cozmo14047 You can read and restore/write your statistics
- Read EEPROM_FILAMENTUSED : `D3 Ax0ff1 C4`
- Read EEPROM_TOTALTIME : `D3 Ax0fed C4` 
- Write EEPROM_FILAMENTUSED : `D3 Ax0ff1 X<hex value in lowercase a-f>`
- Write EEPROM_TOTALTIME : `D3 Ax0fed X<hex value in lowercase a-f` 

Example:
Read 
```
Send: M78
Recv: STATS 74 min 193 cm.
....
Send: D3 Ax0ff1 C4
Recv: D3 - Read/Write EEPROM
Recv: 000ff1  c1 00 00 00
...
Send: D3 Ax0fed C4
Recv: D3 - Read/Write EEPROM
Recv: 000fed  4a 00 00 00
....
```
Restore/Write same values
:exclamation: !!! ATTENTION that the hex values need to be **lowercase `a-f`** while `A and X` need to be UPPERCASE AND the hex values NEED to be 2 chars or a multiplier of 2 with maximum of 16 chars in one D3 command !!! :exclamation:  
```
D3 Ax0ff1 Xc1000000
D3 Ax0fed X4a000000
```

cozmo14047, 2023-11-29:
> @cozmo14047 You can read and restore/write your statistics
> 
> * Read EEPROM_FILAMENTUSED : `D3 Ax0ff1 C4`
> * Read EEPROM_TOTALTIME : `D3 Ax0fed C4`
> * Write EEPROM_FILAMENTUSED : `D3 Ax0ff1 X<hex value in lowercase a-f>`
> * Write EEPROM_TOTALTIME : `D3 Ax0fed X<hex value in lowercase a-f`
> 
> Example: Read
> 
> ```
> Send: M78
> Recv: STATS 74 min 193 cm.
> ....
> Send: D3 Ax0ff1 C4
> Recv: D3 - Read/Write EEPROM
> Recv: 000ff1  c1 00 00 00
> ...
> Send: D3 Ax0fed C4
> Recv: D3 - Read/Write EEPROM
> Recv: 000fed  4a 00 00 00
> ....
> ```
> 
> Restore/Write same values ❗ !!! ATTENTION that the hex values need to be **lowercase `a-f`** while `A and X` need to be UPPERCASE AND the hex values NEED to be 2 chars or a multiplier of 2 with maximum of 16 chars in one D3 command !!! ❗
> 
> ```
> D3 Ax0ff1 Xc1000000
> D3 Ax0fed X4a000000
> ```

Ah I see now sorry I didn't get it the first time

3d-gussner, 2023-12-11:
Please consider to close this issue.

closed 2023-12-11 by 4rdful