prusa3d/Prusa-Firmware #4513: [BUG]FW crash detected and a boot loop

numo68, 2023-11-26:
**Printer type** - MK3S upgraded from MK2S
**Printer firmware version** -  3.13.2

**MMU upgrade** - none
**MMU upgrade firmware version** - 

**SD card or USB/Octoprint**

Normally USB/Octoprint but no cable nor card here

**Describe the bug**

After flashing the 3.13.2 the printer entered a boot loop with the "FW crash detected" report. Sometimes the main screen appeared, but on the next start the error message was back. The factory reset followed by a reflash and another factory reset did not change anything. I have seen the message once or twice on the previous 3.13, I don't know whether it was .0 or .1 anymore.

I have flashed 3.12.2 back and all seems fine. Flashed 3.13.2 again, the problem reappeared, 3.12.2 and all is fine.

The crash happens without any SD card nor USB cable plugged in.

**To Reproduce**

Flash the 3.13.2 firmware.
  
**Expected behavior**

The printer boots normally.

gudnimg, 2023-11-26:
Is it possible to get the serial logs from your printer? It may reveal how far into the boot-up it gets, maybe the firmware is getting stuck somewhere.

> Sometimes the main screen appeared, but on the next start the error message was back.

This seems suspicious to me.

If 3.12.2 works fine then we can probably rule out this is not a bootloader issue.

numo68, 2023-11-26:
> Is it possible to get the serial logs from your printer? It may reveal how far into the boot-up it gets, maybe the firmware is getting stuck somewhere. `"FW crash detected"` message is also used for watchdog timeouts.

How do I get the logs? Frankly, I just finished to calibrate everything after the factory reset, so I'll only experiment with this if the risk of losing the settings is minimal and the probability of providing useful information is high.

Yep, the bootloader seems fine, also the flashing including verify was always successful on the first attempt in case it matters, so the board itself does not seem to be damaged, at least not severely.

Were there different board versions where the difference could matter over time? Mine is reported as Einsy_10a in the support menu.

pdelagrave, 2023-12-07:
I have the same issue.
Printer is an original MK3S+ I bought as a kit. It worked fine for 2 years until I tried the newest firmware. Having the issue you're describing on any firmware larger than 3.12.1.

Contacted support and they told me to disconnect everything from the Einsy board and reconnect one cable at time to find the source of the problem. Just tried that and the problem still happens with no cable other than power connected.

numo68, 2023-12-08:
Update: it _sometimes_ also happens with 3.12.2, but maybe one of ten starts and I cannot exclude that it is related to the Octoprint starting simultaneously, resetting the board when the USB connection is established. With 3.13.2 it however happens with nothing connected at all.

numo68, 2023-12-08:
@gudnimg 

> Is it possible to get the serial logs from your printer? It may reveal how far into the boot-up it gets, maybe the firmware is getting stuck somewhere. `"FW crash detected"` message is also used for watchdog timeouts.

Alright:
```
start
echo: 3.13.2-7080
SpoolJoin is Off
echo: Last Updated: Oct 20 2023 17:49:53 | Author: (none, default config)
echo: Free Memory: 2513  PlannerBufferBytes: 1760
echo:Stored settings retrieved
adc_init
Hotend fan type: NOCTUA
CrashDetect DISABLED
echo:SD init fail
start
echo: 3.13.2-7080
SpoolJoin is Off
echo: Last Updated: Oct 20 2023 17:49:53 | Author: (none, default config)
echo: Free Memory: 2513  PlannerBufferBytes: 1760
echo:Stored settings retrieved
adc_init
Hotend fan type: NOCTUA
CrashDetect DISABLED
echo:SD init fail
// action:dump_available
echo:busy: paused for user
```
At this point the FW crash report is displayed.

Inserting the SD card (I have no need for one as I am using Octoprint exclusively) resolves the situation, so the crash seems to somehow happen while probing it.

pdelagrave, 2023-12-27:
Exact same problem with my MK3S+. Inserting an SD card also worked as a workaround.

numo68, 2024-10-09:
upgraded to mk3.5 in  the meantime so not relevant for me anymore, nor am I able to test, but if it was not fixed, it probably should.

closed 2024-12-17 by github-actions[bot]