prusa3d/Prusa-Firmware #4517: [ENHANCEMENT]<SD card failed>

ah1102, 2023-11-29:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MK3S

**Is your enhancement related to a problem? Please describe.**
My SD card has failed. And for a long time I could not detect this problem. The printer simply rebooted and the print was reset. This happened several times. It's as if the printer's main board has failed.
  
**Describe the solution you'd like**
  Add code that will detect if the SD card is damaged. What happens when reading from a memory card. And the user will be shown that the memory card is damaged.

3d-gussner, 2023-11-30:
@ah1102 Thanks for the idea, we have a look into that if it is possible.
Can you please describe your findings bit more detailed
- When does it happen? After inserting the SD or while printing at some random location?
- Did you check that the SD card slot is working well? What happens when you wiggle the SD card a bit?
- How did you determined that the SD card is actually damaged? Which program did you use to check your SD card?
- Did you try to copy the data from the SD card to PC and check if the files are complete?
  - Copy one very very large file from PC to SD card
  - Rename the PC filename
  - Copy that file from SD card to
  - Compare the original and from SD card copied files
- Did you try to reformat the SD card, does it fail to format.

3d-gussner, 2023-11-30:
@ah1102 Thanks for the answers. We will see what is possible but can't promise anything and how long this may take.
I keep it assigned to myself to not loose track of this.