prusa3d/Prusa-Firmware #4521: [BUG]Unknown Error / Unable to Show Details

Jam-ehz, 2023-11-30:
Prusa Mini+
5.1 Nov 23 Stable FW

Using OctoPrint

Started a print job overnight, checking it in the morning the print was successful, but it was constantly beeping with this error screen:

[image]

Unknown Error, and Unable to show deta(ils?)

Haven't been able to reproduce it, but here's the error dump file and gcode that was used:
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware/files/13515300/dump_buddy.zip)

Not very helpful with an error like this, but might as well throw it out there in case someone else has it!

3d-gussner, 2023-11-30:
@Jam-ehz This is the repository for Prusa MK2.5/S MK3/S/+ please open the ticket for MINI, MK4, XL in Prusa-Buddy-Firmware https://github.com/prusa3d/Prusa-Firmware-Buddy

And please close it here.

Jam-ehz, 2023-11-30:
Ah, Sorry!

closed 2023-11-30 by Jam-ehz