prusa3d/Prusa-Firmware #4531: [FEATURE REQUEST]<Move "Disable steppers" to main menu while not printing>

Pitkox, 2023-12-06:
**Printer type** -MK3S
  
When we change the filament, there is always a problem with leaking residue (pressure in the hotend). there are many memes where people stand over the printer at the beginning of printing with tweezers to remove this "hair" off of the nozzle. You can get rid of it by slightly pulling the filament away from the hotend. I always do this and it helps a lot to keep the nozzle clean before printing. To do this, you need to turn off the stepper motors that hold the filament after it is loaded. Wouldn't it be easier if we gave this option at the top, to the main menu only when we prepare the printer for printing, or automatically turn off the stepper motors if the printer is not printing. I understand that you can't give this option while printing, but it's different, when printing it's also not in the settings. Please consider this, I've written about this before, but the problem hasn't magically disappeared, and I don't believe I'm the only one with this problem...

sarusani, 2023-12-06:
I used to be one of the people standing over the printer with tweezers at the beginning of every print. :)

I made a change to my firmware to retrace 10mm after the filament loading. (Instead of doing it manually)
Solved this issue completely. I could make a PR if that's something people are interested in... But it's not on me to decide if it will be included.

Pitkox, 2023-12-07:
Yeah this idea might be helpful too but I think my case will be more universal, Maybe good idea is to implement both and this retraction thing set like option (ON-OFF)?

3d-gussner, 2023-12-09:
@sarusani If you create a PR with good arguments and some test scenarios and results we will gladly look into this and consider to approve the PR in future releases (not in FW 3.14.0)
@Pitkox Moving the "Disable Steppers" to main menu will probably not approved as it changes the main menu and most users don't use "Disable Steppers" at all.

Pitkox, 2023-12-09:
It's not about using or not. It's about the problem that this could fix. I'm one and only with this leak problem? I don't think so....
At least, think about @sarusani idea but as I mentioned this approche is not universal.

closed 2024-02-15 by github-actions[bot]