prusa3d/Prusa-Firmware #4532: [ENHANCEMENT] Faster preheat. M104 M109 M140 M190 enhacement

KOBALT3D, 2023-12-06:
**Printer type** - MK3S, possibly all Prusa models

Preheating process is sometimes long (using LCD menu or starting g-code) because uses PID method. It also causes oozing until temperatures are stable.
  
I am suggesting to add another parameter to all heating commands, **P** (as full Power) for example to temporarily disable PID heating or using maximum power until temperature reaches specified value.

Panayiotis-git, 2023-12-06:
Isn't PID regulating the temperature with the exact same logic? As long as the deviation of the current temperature to the set temperature is large enough, it uses the maximum power to heat the element.

Have you calibrated the PID controllers of your printer?

KOBALT3D, 2023-12-06:
PID also reduces (cycles) power before temperature reaches specified value. This will slow the heating process. If you preheating from room temperature to certain temperature there is no need to slow down the heating process or to reach precisely certain temperature. The temperature will be stabilized relatively fast during Mesh Bed Leveling.

closed 2024-02-21 by github-actions[bot]