prusa3d/Prusa-Firmware #4536: Power Panic failed to recover

jps4you, 2023-12-10:
### Description of the bug

Firmware 3.13.2 (3.0.1) on Prusa MK3s+
System failed to recover after a power failure. System only booted up into main page. 

### Project file & How to reproduce

[Layer-3.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13627965/Layer-3.3mf.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.7.0+MacOS-x64

### Operating system

MacOS 13.6.1 

### Printer model

Prusa MK3S+

gudnimg, 2023-12-10:
Are you using Octoprint/PrusaLink or any sort of remote host?

jps4you, 2023-12-10:
sorry, should've mentioned that. no, straight from SD card.

gudnimg, 2023-12-10:
Thanks, that’s good to know. I have a few more questions :)

How far into the print did the power panic occur? Was it at the first few minutes or multiple hours into the print?

How much time are we talking about between the power panic happening and when you got to the printer again to recover the print? Enough time for the nozzle to fully cool down?

Is there a possibility you had multiple power panics consequtively? It should not cause any issue, but I wouldnt rule it out.

jps4you, 2023-12-10:
the print was about 2 hours into a 7 hour print. It was one of those over night print that after confirming first few layers were good and went to bed. I found the printer at the initial screen as if it had completed booting up. The bed and nozzle were cold, nozzle is stuck to the print at the layer when power panic occurred. I can only confirmed I had a power failure from my microwave. However, I don't know if there were multiple power failures during the night. The printer is plugged into a surge protector as normal. I've successfully tested power panic feature when I first got the printer back in 2020, but luckily never had the chance to use it until today. Sadly, system failed to recover :) Though, I guess was a good thing. The organic support starting in mid air issue (already reported) was causing quite the ugly print as free floating extruded filament was stuck to outer walls all over the project about an hour into the print.

gudnimg, 2023-12-10:
> I found the printer at the initial screen as if it had completed booting up. The bed and nozzle were cold, nozzle is stuck to the print at the layer when power panic occurred. 

Hmm to me this sounds like the printer had automatically tried to recover from the power outage (which means power came on again before the nozzle got below 175C). But then something happens which causes the printer to stop. 

When power panic triggers, the extruder is _always_ moved to the side. The fact that the nozzle was on the print indicates there was some recovery attempted. But why it would stop there (on the print) is a mystery 🤔

I can imagine multiple short power outages could be an issue if they happen at the right time but thats just speculation.

cozmo14047, 2023-12-29:
Thought I'd mention this, sometimes if your connected to a surge protector the power can go out slowly. What that means is its not on/off, its like the power slowly gets less and less volts which can not trigger the power panic as it still gets powerd but not enough for the printer to run. I've found the power panic only works when the power goes on/off. awhile ago I belevie there was a thread of someone who had the same issue I'd recomend testing it without or with another surge protector. Regards

3d-gussner, 2024-07-08:
Closing due to lack of interaction.

closed 2024-07-08 by 3d-gussner