prusa3d/Prusa-Firmware #4537: [BUG] Error compiling firmware with python

seppelicous, 2023-12-11:
**Describe the bug**
  I downloaded the latest source file via clone git. 
  I ran the compile command as seen in below quote and always get the same error.
  It happens on MacOS Sonoma (Mac M1) and on Windows 11
Actually I don't know which informations are helpful for you, so maybe the error output is acutely helpful enough.
`Prusa git:(master) ✗ python utils/build.py --preset mk4 --generate-bbf 
Switching to Buddy's virtual environment.
You can disable this by setting the BUDDY_NO_VIRTUALENV=1 env. variable.
Traceback (most recent call last):
  File "/Users/Sebastian/Prusa/utils/build.py", line 832, in <module>
    main()
  File "/Users/Sebastian/Prusa/utils/build.py", line 623, in main
    bootstrap.switch_to_venv_if_nedded()
  File "/Users/Sebastian/Prusa/utils/bootstrap.py", line 211, in switch_to_venv_if_nedded
    os.execv(str(venv_bin_dir / 'python'),
FileNotFoundError: [Errno 2] No such file or directory`

3d-gussner, 2023-12-12:
@seppelicous This is the repository for the 8-bit MK2.5/S MK3/S/+ printers please close the issue here and open it in the 32-bit MINI, MK4, XL repository https://github.com/prusa3d/Prusa-Firmware-Buddy

seppelicous, 2023-12-12:
Sorry, wrong repository.

closed 2023-12-12 by seppelicous

3d-gussner, 2023-12-12:
@seppelicous No Problem, hope they can help you in the 32-bit repo.