prusa3d/Prusa-Firmware #4538: this srkypt does not install anything, so the next steps do not work (missing cmake/ninja)

Garomek, 2023-12-12:
# automatically setup dependencies
./utils/bootstrap.py

# configure and build
mkdir build
cd build
cmake .. -G Ninja -DCMAKE_BUILD_TYPE=Release -DCMAKE_TOOLCHAIN_FILE=../cmake/AvrGcc.cmake
ninja
zsh: command not found: #
Klonowanie do „Prusa-Firmware”...
remote: Enumerating objects: 49291, done.
remote: Counting objects: 100% (1814/1814), done.
remote: Compressing objects: 100% (570/570), done.
remote: Total 49291 (delta 1335), reused 1602 (delta 1244), pack-reused 47477
Pobieranie obiektów: 100% (49291/49291), 51.32 MiB | 2.18 MiB/s, gotowe.
Rozwiązywanie delt: 100% (37422/37422), gotowe.
zsh: command not found: #
Downloading ninja-1.10.2
Extracting ninja-1.10.2
Downloading cmake-3.22.5
Extracting cmake-3.22.5
Downloading avr-gcc-7.3.0
Extracting avr-gcc-7.3.0
Downloading prusa3dboards-1.0.6
Extracting prusa3dboards-1.0.6
zsh: command not found: #
zsh: command not found: cmake
zsh: command not found: ninja
romek@macbook-pro-1 build % 'cd..'
zsh: command not found: cd..
romek@macbook-pro-1 build % gettext
gettext: brakujące argumenty

3d-gussner, 2023-12-12:
Did you follow the steps one by one?
Do you have a macbook? Which OS do you have installed? 

# clone the repository

1. `git clone https://github.com/prusa3d/Prusa-Firmware`
2. `cd Prusa-Firmware`

# automatically setup dependencies
3. `./utils/bootstrap.py`

# configure and build
4. `mkdir build`
5.` cd build`
6. `cmake .. -G Ninja -DCMAKE_BUILD_TYPE=Release -DCMAKE_TOOLCHAIN_FILE=../cmake/AvrGcc.cmake`
7. `ninja`

Garomek, 2023-12-12:
Hi. I did it step by step. No cmake no ninja

￼
Pozdrawiam / Best Regards

Roman Gaik
***@***.***

Garomek, 2023-12-12:
Yes. I have MacBook Pro with Sonoma 14.2.
My steps You can watch on the top.

3d-gussner, 2023-12-13:
I may have found the issue.
We as developers have cmake, ninja installed on our computers and OS and you probably not so it has not been found.

Please try this. I am not sure if it will work/compile on your MacBook Pro with Sonoma 14.2 as we don't have any Macs to test this. 

Also ensure: 
Python >= 3.6 is also required with the following modules:

    pyelftools (package python3-pyelftools)
    polib (package python3-polib)
    regex (package python3-regex)

Additionally gettext is required for translators.

# clone the repository

1. `git clone https://github.com/prusa3d/Prusa-Firmware`
2. `cd Prusa-Firmware`

# automatically setup dependencies

3. `./utils/bootstrap.py`

# check if dependecies are working

4. `../.dependencies/cmake-3.22.5/bin/cmake --version` should show `cmake version 3.22.5` or similar
5. `../.dependencies/ninja-1.10.2/ninja --version` should show `1.10.2` or similar

# configure and build
6. `mkdir build`
7. `cd build`
8. `../.dependencies/cmake-3.22.5/bin/cmake .. -G Ninja -DCMAKE_BUILD_TYPE=Release -DCMAKE_TOOLCHAIN_FILE=../cmake/AvrGcc.cmake`
9. `../.dependencies/ninja-1.10.2/ninja`

3d-gussner, 2023-12-13:
can you change to the folder Prusa-Firmware/.dependecies/cmake-3.22.5/bin and is there a cmake executable?
in this folder try cmake --version to see if that works.

BTW your screenshot has not been added

Garomek, 2023-12-13:
Please find a screenshot attached.
[image]

Garomek, 2023-12-13:
Please write me step by step lines to enter in the console of my MacBook. I'm not a programmer so maybe I'm typing something wrong

3d-gussner, 2023-12-15:
@Garomek I don't know MacOS and can't write exact step by step guide except I have done above.
Try to change to the  .dependencies/cmake-3.22.5/bin/cmake folder with multiple `cd ` commands and run `cmake --version`

closed 2024-02-26 by github-actions[bot]