prusa3d/Prusa-Firmware #4540: [BUG]

Cbokar, 2023-12-15:
After some time of printing, 1 hour or so, there are no more temperatures shown under: settings— info —sensors: heartbreak and boardtemperature are shown as: not initialized. After a reset the temps are shown properly again.
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.
-->
**Printer type** - Mk4, factory assembled. FW 5.1.0.  slicer 2.7.

No matter what gcode file, gcode or bgcode

Cbokar, 2023-12-15:
[image]

3d-gussner, 2023-12-18:
@Cbokar This is the repository for the MK2.5/S and  MK3/S/+ please reopen this issue https://github.com/prusa3d/Prusa-Firmware-Buddy and close it here.

3d-gussner, 2023-12-20:
@Prusa-Support Please move this issue to buddy firmware

3d-gussner, 2023-12-28:
@Cbokar You opened the issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3626 but forgot to close it here.
Closing it.

closed 2023-12-28 by 3d-gussner