prusa3d/Prusa-Firmware #4542: [BUG] MMU2S Filament load LED not operating

plug-it-in, 2023-12-16:
**Printer type** - MK3S
**Printer firmware version** -  3.13.2-7080

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 3.0.1

**SD card 

**Describe the bug**
Since the 3.0.1 update on the MMU2S, the LED does not light up green on the MMU2S when a filament is loaded. This is a regression in function and is critical feedback for the end user. I spoke with support and they explained it away that the rewrite of the code had changed this and was not necessary as any errors would be reported on the LED. This visual positive feedback is essential especially when looking for loading issues. I note that when the filament feeds to the nozzle the LED comes on then (kind of after the fact event). Please reinstate the green LED function as it was originally.
  
**To Reproduce**
 Load filaments on previous release and compare
  
**Expected behavior**
  When the filament is preloaded to the MMU the green LED should come on in the position its loaded into.

HebertGB, 2023-12-27:
Same here. 

Printer MK3S firmware 3.13.2-7080
MMU2S firmware 3.0.1

plug-it-in, 2024-02-26:
What? This needs foxing

3d-gussner, 2024-03-08:
See https://github.com/prusa3d/Prusa-Firmware/issues/4289

closed 2024-03-08 by 3d-gussner