prusa3d/Prusa-Firmware #4544: Firmware bugs out when trying to print certain OBJ files

Dmalin3D, 2023-12-18:
**Printer type** - MK4
**Printer firmware version** - 5.1.2

**SD card or USB/Octoprint**
USB Drive

**Describe the bug**
When printing a OBJ with mutliple colors and a tool change between each color the screen goes blank when you press print (the time and date or w/e is still there but the rest is black). If I revert to the 5.0.0. Firmware it prints fine
  
**To Reproduce**
Try to print this 3mf
https://drive.google.com/file/d/1o_Frxv_obycrUB4JjUKvbszUhyORSPVz/view?usp=sharing
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.
gcode file on attached google drive link
https://drive.google.com/file/d/1eWKwu8L52aaGpylit99Z1Kh3-EZpao2B/view?usp=sharing

3d-gussner, 2023-12-19:
@Dmalin3D This is the MK2.5/S and MK3 repository. Please close this issue here and reopen it in the MK4, MINI, XL repository https://github.com/prusa3d/Prusa-Firmware-Buddy

Dmalin3D, 2023-12-19:
Moved to another repository

3d-gussner, 2023-12-20:
@Dmalin3D 
>Moved to another repository

Thanks I will close the issue here.

closed 2023-12-20 by 3d-gussner