prusa3d/Prusa-Firmware #4550: [BUG] Prusa XL reports incorrect filament type

OneEyeLeft, 2023-12-28:
**Printer type** - XL
**Printer firmware version** -  5.2.1

**SD card or USB/Octoprint**
USB

**Describe the bug**
When starting new prints on a 5 tool XL that have been loaded to the USB drive via Prusa Link the filament type for some tool heads seems to change without user interaction or a recent filament change. Example: I have the following set up right now:
Tool1 - .4mm nozzle - PLA
Tool2 - .4mm nozzle - PLA
Tool3 - .4mm nozzle - PETG
Tool4 - .4mm nozzle - PLA
Tool5 - .6mm nozzle - PLA
In Prusa Slicer I slice my model for Tool1, .4mm, PLA. When I transfer the bgcode to the printer and select it the printer reports that the current filament type for Tool 1 is PETG, not PLA. I have not changed filaments for a while and have printed with several tools, including tool 1 over the last few days. I have to select Tool 1 and choose to accept the warning that "Tool 1 has the incorrect filament".

**To Reproduce**
I do not have steps to reproduce but I realized this is the second time this has happened to me. I do print .6mm, .4mm sliced objects and do print PLA, PETG, and TPU. The only action I think that was possibly related was a filament change on Tool 5 from PETG to PLA a day ago but I have since printed on Tool 5 and Tool 3 with no issue. 
  
**Expected behavior**
Prusa XL will retain knowledge of its loaded filament type between prints and jobs, especially when the filament has not changed. 
  
**G-code**
No gcode  relevant to this bug
**Video**
  Please attach a video. It usually helps to solve the problem.
Please note in the photo attached that only tool 3 had PETG loaded when this error occurred. 
[image]

3d-gussner, 2023-12-28:
@OneEyeLeft This is the issue tracker for the Prusa MK2.5/S and MK3/S/+ 8-bit printers. Please close this issue here and open it in the MINI, MK4, XL repository https://github.com/prusa3d/Prusa-Firmware-Buddy.

OneEyeLeft, 2023-12-29:
Done. Opened as https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3647

closed 2023-12-29 by OneEyeLeft