prusa3d/Prusa-Firmware #4557: Since I installed 5.1.2 Firmware on my MK4 I cannot see what I am printing on the LCD screen

prusajb, 2024-01-02:
Since I installed 5.1.2 Firmware for Original Prusa MK4, I cannot see the thumbnail image of the part I am going to print on the LCD screen. I can't see it before starting printing or during printing.

does anybody know how I can fix it? When you have many files on the USB it is helpful to view the file before starting the printing process.
[image]

JonMrowczynski, 2024-01-03:
Hello prusajb, I hope that you have been doing well! 

It appears that your issue is for the MK4. The repository for this can be found [here](https://github.com/prusa3d/Prusa-Firmware-Buddy). As stated in the README for this [repository](https://github.com/prusa3d/Prusa-Firmware?tab=readme-ov-file#prusa-firmware-mk3) "This repository contains the source code and the development versions of the firmware running on the [Original Prusa i3](https://prusa3d.com/) MK3S/MK3/MK2.5S/MK2.5 line of printers."

From my understanding of what is usually done in this case, it would be most helpful for you to close this issue here and repost it on the [issues for the MK4 firmware](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues).

3d-gussner, 2024-01-09:
@Prusa-Support Please move this Issue to Buddy firmware

3d-gussner, 2024-02-07:
Closing issue as it is for the buddy firmware

closed 2024-02-07 by 3d-gussner