prusa3d/Prusa-Firmware #4558: [BUG] MMU Idler cannot home message leads to unrecoverable state

PxT, 2024-01-04:
**Printer type** - MK3S
**Printer firmware version** -  3.13.2

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 3.13.2

**SD card or USB/Octoprint**
 SD

**Describe the bug**
Upon receiving an “IDLER CANNOT HOME” error from the mmu there is an option to adjust the sensitivity. Going into this menu then back by selecting “Done” returns the printer to the main info screen with “OK” on the bottom line and the print does not resume. Both temperatures remain set but otherwise the printer does not appear to be actively doing anything. Pause / Stop print options do not appear in the menu and there’s no apparent way to resume or recover the print. I have experienced this more than once, with different gcodes. Not sure how to specifically reproduce as I haven’t tried to artificially trigger the idler error message.

gudnimg, 2024-01-05:
> Going into this menu then back by selecting “Done” returns the printer to the main info screen with “OK” on the bottom line and the print does not resume. 

The idler error screen should have been rendered again to give you the “Retry” button. Can you confirm this didnt happen?

This can explain your scenario since the MMU will stay in an error state waiting for the “retry” command.

I have never seen this when the printer is idle. Maybe something strange happens when printing or the MMU is running a toolchange.

PxT, 2024-01-05:
Confirmed, this didn't happen.

gudnimg, 2024-01-06:
@PxT I can confirm the issue and can reproduce it quite easily. The issue seems to happen when the Idler fails to engage/home during toolchange. The printer tries to tell the MMU to unload which the MMU rejects since it think it is doing a toolchange.

I will look more into this.

sithlord48, 2024-05-01:
+1 just had this happen. tool change got held up i mis clicked into tune yeah there is no way to get back .. lost about 1kg of filiment and 18ish hours worth of the mmu job i was printing..  :(

gudnimg, 2024-05-04:
@sithlord48 @PxT I've submitted another PR which does at least mitigate the issue by making the Tune button not do anything if the MMU is in the middle of a toolchange or another command. This way this issue will at least stop ruining prints.

To make the Tune button work while printing is a whole other matter. But the important thing is it should not ruin prints!

closed 2024-05-06 by 3d-gussner