prusa3d/Prusa-Firmware #4564: MMU2S FW 3.0.1 TMC driver error

duebelpapst, 2024-01-10:
Printer type - MK3 w/ upgrade Kit to MK3S
Printer firmware version - 3.13.2

MMU upgrade - MMU2S
MMU upgrade firmware version - 3.0.1

SD card or USB/Octoprint
I print via SD card

Describe the bug
Since I updated my MK3S to 3.13.2 and the MMU2S to 3.0.1, I get an error message from the MMU almost every time I print.
Usually only at the end of the print, but sometimes during the print while the print is still running.
These are always prints with only one filament.
The error message is "TMC driver error prusa.io/04321 FI:1 FS:0"
In my opinion, all steppers of the MMU move at twice the speed since the update, can it be related to this?

To Reproduce
It happens to almost every print and gcode

Expected behavior
A clear and concise description of what you expected to happen.

G-code
G-code file is not accepted for upload, how can I upload the code.

Video
[image]

3d-gussner, 2024-01-10:
@duebelpapst Thanks for the report, sorry to hear that you have that issue quite often.
As you can see in the [prusa.io/04321](https://prusa.io/04321) article the MMU3 PD board stabilizes the power to the MMU controller board.
The new firmware "monitors" the MMU and also TMC drivers way better than the "old" MMU2 firmware.

- Which PSU do you have Silver or Black
- Which Voltage are shown on LCD Support -> Voltages
- Please check your cables and the pins from the EINSY P3 connector to MMU controller
  - "Bad/Loose" connections can cause additional issues.

Please consider to [upgrade to the MMU3](https://www.prusa3d.com/product/original-prusa-mmu2s-to-mmu3-upgrade-for-mk3s-3/) as this will improve the MMU significantly.

duebelpapst, 2024-01-11:
> @duebelpapst Thanks for the report, sorry to hear that you have that issue quite often. As you can see in the [prusa.io/04321](https://prusa.io/04321) article the MMU3 PD board stabilizes the power [...]
     **it is silver**
> * Which Voltage are shown on LCD Support -> Voltages
     **23,9V / 23,9V / 5,0V**
> * Please check your cables and the pins from the EINSY P3 connector to MMU controller
>   
>   * "Bad/Loose" connections can cause additional issues.
   **all good**
> 
> Please consider to [upgrade to the MMU3](https://www.prusa3d.com/product/original-prusa-mmu2s-to-mmu3-upgrade-for-mk3s-3/) as this will improve the MMU significantly.

Well, you make a questionable firmware update and I have to spend 150-250€ to get it working again, I don't find that very funny.

Could it be that your software people have set the current of the MMU motors higher, that would also explain why the error occurs without moving the motors, because they also need current to keep the motors in position.
Is it not possible to change this again, or at least make it adjustable via software?
I also had error #04306, which is also an error that was supposedly caused by the power supply.
Before the update I didn't have a single error message in 4 years.

I have now increased the voltage on the power supply from 23.9V to 24.5V.
The 5.0V has remained at 5.0V.

Regards,
Stefan

duebelpapst, 2024-01-14:
Hello Hello everyone,

it seems that the solution was simply to adjust the voltage on the power supply from 23.9V to 24.5V, since then the MMU2S has been working perfectly.

Regards,
Stefan

3d-gussner, 2024-01-15:
> Hello Hello everyone,
> 
> it seems that the solution was simply to adjust the voltage on the power supply from 23.9V to 24.5V, since then the MMU2S has been working perfectly.
> 
> Regards, Stefan

Glad to hear you got it working. Please consider to close this issue as users still can find it even it is closed.

closed 2024-01-15 by duebelpapst