prusa3d/Prusa-Firmware #4568: [BUG] XL hangs / pauses mid print causing blobs

averypierce, 2024-01-13:
**Printer type** - XL 5T
**Printer firmware version** -  5.1.2, and 5.1.2 again after re-flash

**SD card or USB/Octoprint**
  Stock USB, alternate USB, Stock USB after re-formatting

**Describe the bug**
During multitool prints, XL randomly hangs, causing blobs. Never seems to be in the same spot. It always recovers. No errors on screen. Temps normal. Confirmed on tools 1, 2, and 5.

No issues with single tool prints. 
  
Found a report of what sounds like same issue on forum https://forum.prusa3d.com/forum/original-prusa-xl-tool-changer-general-discussion-announcements-and-releases/xl-pauses-does-nothing-for-over-10-seconds-and-oozes-all-over-the-print

**To Reproduce**
Start a multi tool print and wait til somewhere mid print. I usually send my prints straight from PrusaSlicer upload over PrusaLink, but it is the same if starting directly from USB. Occurs reliably on multicolor benchy.  
  
**Expected behavior**
I expect tool head to keep moving 

**G-code**
[3DBenchy - Multi-part - Single - Bridge roof - (3DBenchy_0.4n_0.2mm_PLA,PLA,PLA,PLA_XLIS_1h33m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/13929243/3DBenchy.-.Multi-part.-.Single.-.Bridge.roof.-.3DBenchy_0.4n_0.2mm_PLA.PLA.PLA.PLA_XLIS_1h33m.gcode.zip)

**Video**
[  Please attach a video. It usually helps to solve the problem.](https://www.youtube.com/shorts/8cQEBd_XI7c)

JonMrowczynski, 2024-01-14:
Hello averypierce, I hope that you have been doing well!

It appears that your issue is for the XL. The repository for this can be found [here](https://github.com/prusa3d/Prusa-Firmware-Buddy). As stated in the README for [this repository](https://github.com/prusa3d/Prusa-Firmware?tab=readme-ov-file#prusa-firmware-mk3) "This repository contains the source code and the development versions of the firmware running on the [Original Prusa i3](https://prusa3d.com/) MK3S/MK3/MK2.5S/MK2.5 line of printers."

From my understanding of what is usually done in this case, it would be most helpful for you to close this issue here and repost it on the [XL firmware repository's issue tracker](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues) so that it gets seen by the right people.

averypierce, 2024-01-14:
d'oh!

closed 2024-01-14 by averypierce