prusa3d/Prusa-Firmware #4570: XYZ Calibration fails if INVERT_Y_DIR is set to 1

fitch22, 2024-01-16:
Printer type MK3S with mods
Printer firmware 3.13.2
Operation from LCD

Describe the bug:
I have a mostly MK3S printer.  It started life as scratch built i3 but upgraded with kits from Prusa to MK2, then MK2.5 and finally MK3S.  Other than my X, Y, and Z limits are ever so slightly different than a Prusa MK3S, my Y motor is mounted backwards.

I can correct for the Y motor mount by either 1) change INVERT_Y_DIR to 1 in MK3S.h or 2) reverse the plug connection on the einsy board.

Changing INVERT_Y_DIR to 1 corrects the motion, and the printer can home, print, do everything it needs to do except XYZ calibration.  It appears that the Y direction is assumed in the calibration algorithm and it only works if INVERT_Y_DIR is a 0.  For me, the solution was #2 above, I had to reverse the wires.

To Reproduce
If you have a stock Prusa MK3S, you can duplicate this issue by reversing the connection of the Y motor and setting INVERT_Y_DIR to 1 in MK3S.h.  Then attempt an XYZ calibration.

You can observe that the calibration starts at the first point and spirals around the sensor location as expected.  But when it starts to sweep back and forth to get exact location of the sensor, it starts at the sensor and moves off.  Then for the 2nd sensor location it is lost and does not even spiral around the sensor.  Finally for sensors 3 and 4 it is so far off that it bumps into the Y limits.  Obviously the calibration fails.

The algorithm is very complicated to the casual observer and I was unable to figure out how to correct it.  But the firmware engineer in charge of this can probably fix it in a few minutes.

3d-gussner, 2024-01-17:
@fitch22 Thanks for the bug report, but we merged https://github.com/prusa3d/Prusa-Firmware/pull/1263 quite long time ago and it should have fixed the issue.

fitch22, 2024-01-17:
I am building using cmake and vscode on windows.  I copied and created my own header file for the variant.  Maybe the cmake stuff does not include the new header properly in sm4.h.  I don’t read/write cmake so this is just a guess.

fitch22, 2024-01-17:
More info.  I did a clean build of 3.12.2.  It has the same behavior.  For this, my build environment was WSL using PF_build.sh.  So the bug does not appear to be related to the new cmake stuff.

fitch22, 2024-01-17:
It looks like the merge of #1263 was done on 3.10.  Not sure when the new XYZ calibration came into the mix, but is it possible that the new calibration re-introduced this same bug?

fitch22, 2024-03-27:
Why was this not fixed for 3.13.3?

3d-gussner, 2024-03-27:
@fitch22  FW 3.13.3 was mainly a bug fix for Ghost layer shifts and with minimal changes it got tested very fast. Any other changes would have delayed the important release.

sl1pkn07, 2024-07-07:
i think have the same issue in here https://github.com/prusa3d/Prusa-Firmware/issues/4604#issuecomment-2212033011

i'll test switch cables instead invert INVERT_Y_DIR

greetings