prusa3d/Prusa-Firmware #4571: [BUG] xyx callibration fails

Fojtik, 2024-01-17:
**Printer type** - MK3S+
**Printer firmware version** - 3_13_2
**MMU upgrade** - ----
**MMU upgrade firmware version** - ------

**SD card or USB/Octoprint** N/A

**Describe the bug**
After upgrading firmware the printer automatically executes xyz callibration and it fails with a message that is not described here: https://help.prusa3d.com/article/xyz-calibration-error-messages-mk3-mk3s-mk3s_112340

  A clear and concise description of what the bug is.
  
**To Reproduce** Run xyz callibration
 
**Expected behavior** xyz callibration passes
 
**G-code**  N/A

**Video**  https://storage.googleapis.com/prusa3d-content-prod-14e8-wordpress-forum-prod/2024/01/186c850d-20240116_193701.jpg

I did not fiddle with INVERT_Y_DIR, I do not know what is it. May be that this is a duplicity of: https://github.com/prusa3d/Prusa-Firmware/issues/4570 May be not.

3d-gussner, 2024-01-17:
@Fojtik have you tried to contact Prusa Chat support? If that issue isn't solved then please update your github issue here.

Fojtik, 2024-01-17:
Yes, I have contacted Průša support yesterday, no response yet.
I have standard stock Průša printer without any modifications and it asks me to upgrade firmware before wvery print.But this firmware is somehow deffective.

sarusani, 2024-01-17:
I just checked the firmware and the error is thrown when there's a problem with the Z axis. So it's not related to the INVERT_Y_DIR problem.

You mentioned that it's working with older firmware versions. Can I ask you which is the latest one that works for you?

Fojtik, 2024-01-17:
I have these 3 firmwares downloaded:
28.06.2023  13:20         6 659 171 prusa3d_fw_3_12_2_MK3S.zip
06.09.2023  00:04         6 694 867 prusa3d_fw_3_13_1_MK3S.zip
16.01.2024  19:20         6 694 766 prusa3d_fw_3_13_2_MK3S.zip

I am pretty sure that 12.2 is OK. And may be that 13.1. is also OK. Is it safe to flash old firmware?

sarusani, 2024-01-17:
Yes, downgrading the firmware is no problem.
Would be great if you could try 13.1. If that works I can have a look at the differences between the two releases.

Fojtik, 2024-01-19:
Look at this video:
http://78.108.103.11/~fojtik/MK3S/SelhaniKalibraceXYZ.mp4

2 droplets of oil to z axis thread rods fixed a problem. I have blamed a new firmware, but the problem was different. But still the error message is very confusing and printer behavior even more suspicious.
During normal printing the printer uses more power to stepper motors and this problem is hidden.

Fojtik, 2024-01-19:
But there is still one problem of firmware remaining.
When xyz callibration fails from whatever reason, the stepper motor power is not set to usual value.
See this video: http://78.108.103.11/~fojtik/MK3S/20240117_174438.mp4

I cannot reproduce this now ;).

3d-gussner, 2024-01-19:
@Fojtik Please check your bearings and POM nuts. I had once on my X axis similar issues. It turned out that one of the dust-sleeves of a bearing was broken and started to bind more and more in one direction. Removing the sleeve confirmed that it was causing the issue and I replaced the bearing bit later.

Also verify that your Z POM nuts do not bind.
Do you have stock Z-TOP parts or do you have some lead screw constraining one? The lead screw should NOT be constrained at the top!

Thanks for the videos, but the LCD is way too bright to see anything.

>But there is still one problem of firmware remaining.
When xyz callibration fails from whatever reason, the stepper motor power is not set to usual value.
See this video: http://78.108.103.11/~fojtik/MK3S/20240117_174438.mp4

>I cannot reproduce this now ;).

Hard to look for a bug if it can't be reproduced.

Fojtik, 2024-01-19:
**May be that you do not understand me, after 2 droplets of oil xyz test passes OK without any problem.** I have stock parts including sleeves one year old, 25days of continuous print.

The bug could be observed when small friction occurs. Firmware does not return z axis stepper motor's usual current and the current remains reduced.

Fojtik, 2024-02-04:
I has been confirmed deffect of my printer. The printer is still under warranty. This demonstrates printer problem:
https://storage.googleapis.com/prusa3d-content-prod-14e8-wordpress-forum-prod/2024/02/68e29652-nutfriction.mp4
Manufacturer does not recommed using oil.

Undil I fix the issue, I cannot simulate how small friction influences callibration.

Prusa-Support, 2024-03-02:
> I has been confirmed deffect of my printer. The printer is still under warranty. This demonstrates printer problem: https://storage.googleapis.com/prusa3d-content-prod-14e8-wordpress-forum-prod/2024/02/68e29652-nutfriction.mp4 Manufacturer does not recommend using oil.

Thanks for the evidence. That amount of friction is excessive for the trapezoidal nuts to work correctly.
The trapezoidal nuts must spin pretty much frictionless exactly like the first one you spin in the the video.

[Here at step 3](https://help.prusa3d.com/guide/4-z-axis-assembly_168629#168735), the user is requested to remove the trapezoidal nuts. In extremely rare cases over the past many years, there have been cases where the user reported a substantial difference of friction between trapezoidal nuts while removing the trapezoidal nuts from the lead screws as per the assembly step. In those cases, we assumed it was the case of a factory faulty nut and replaced it right away.

In other specific cases, the trapezoidal nuts could get damaged or degrade due to incorrect manipulation.
Here are just a few examples.
- Incorrect assembly/insertion of the X-axis on the lead screws.
- Lubrication. Lubricants are nearly forbidden for this type of trapezoid because they could deteriorate the specific blend of POM material and hinder their "natural" self-lubricating properties - as well as attract debris and build up friction.
- Too long or too short X-axis. This may happen if the smooth rods are of the wrong size, or not inserted all the way inside the X-end printed parts.
- Crooked X-axis. This may happen if a lead screw is skewed or doesn't spin correctly. One of the Z motors could have been incorrectly connected, or one of [the black-printed motor covers](https://cdn.help.prusa3d.com/wp-content/uploads/a204d61ff1214502-800x600.jpg) (or both) is tightened all the way against the motor body, and rubs against the motor, preventing the correct rotation of the lead screw.

_I'm afraid this can't find a solution at the firmware level. The problem is apparently on the hardware side._

Michele Moramarco
Prusa Research

Prusa-Support, 2024-03-06:
I can't exclude a problem is related to issue [#4476](ttps://github.com/prusa3d/Prusa-Firmware/issues/4476#issuecomment-1978997368) so please give it a try with the new firmware release [3.13.3](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.3) and let us know if that fixed the problem.

Michele Moramarco
Prusa Research

Fojtik, 2024-03-16:
I have attempted to reproduce a problem with additional friction with fingers and everything seems to work fine.

The friction of deffective nut must be very high to cause z axis movement to fail.

closed 2024-03-16 by Fojtik

Fojtik, 2024-03-19:
But there was also a firmware problem. After callibration stuck, the stepper mottor did not recover full pover. These rare situations are not tested too much. Currently it seems that printer is doing full reset.

One thing is a nus stuck (but printer printed well to the callibration point) and second thing is wrongly handled situation after such a stuck.

Prusa-Support, 2024-05-18:
I'm sorry, I'm having a hard time trying to understand at the moment and I can't exclude a hardware/assembly problem or something to do with issue https://github.com/prusa3d/Prusa-Firmware/issues/4019.

If the X-axis looks crooked, a Calibrate Z would do the trick.
If the occurrence is recurrent, we may actually be looking at a hardware/assembly problem, ideally to be addressed via individual troubleshooting with our Customer Support.

In any case, if during the troubleshooting a firmware problem can be confirmed, it may not be related to this and the linked issues. In other words, maybe it would deserve a separate and more specific issue.

Michele Moramarco
Prusa Research

Fojtik, 2024-09-22:
I have no problems with my printer now. Once the callibration finishes & passes, the callibration code is not executed again.

I still suspect that callibration does not cover all possible situations (failures), But it never mind.