prusa3d/Prusa-Firmware #4572: [ENHANCEMENT] MK3s update First Layer Calibration to run from front to back

handro123, 2024-01-17:
**Enter what type of printer or upgrade the enhancement applies to.**
MK3s+ with E3d Revo Six
MMU3 UltiMulti

**Is your enhancement related to a problem? Please describe.**
I am a long time Prusa Mini owner and recently got a used MK3s+ with MMU. On the Mini when doing First Layer Calibration it starts at the front and snakes to the back where the square is made in the rear left corner. This is good when running First Layer Calibration as you can easily see the filament, feel with your finger (carefully) and make adjustments as needed. 

With the MK3s+ it starts the First Layer Calibration in the rear and moves forward. You can't easily see the previous rows of extrusion as you have to reach around the X-axis and Extruder. 

  
**Describe the solution you'd like**
Update the MK3s+ First Layer Calibration to match the Prusa Mini / Buddy firmware, where it starts at the front and finishes at back of the print bed.

3d-gussner, 2024-01-18:
@handro123 Thanks for the suggestion. I can't promise we will change it but I will discuss this internally.

pedjas, 2024-04-17:
It may be useful if user can rotate calibration test in 90 degree steps.

3d-gussner, 2024-12-02:
@handro123 Implemented new First layer cal in #4811. If you want to compile the firmware yourself please un-comment `#define NEW_FIRST_LAYER_CAL`  in the variant you need.

Please consider to close this issue.

@pedjas Thanks for the idea. But adding rotation would increase the code size quite a bit, especially when you want to add some LCD menu.

3d-gussner, 2024-12-18:
Closing as it has been implemented for custom builds.

closed 2024-12-18 by 3d-gussner