prusa3d/Prusa-Firmware #4577: [BUG]Multiple stepper motor feedback issues

JeremyDWilliams, 2024-01-23:
**Printer type** -10th. Anniversary (Mk3s+)
**Printer firmware version** - MK3S_MK3S+_3_13_2_7080

**MMU upgrade** -n/a
**MMU upgrade firmware version** -n/a

**SD card or USB/Octoprint**
  SD card

**Describe the bug**
After updating to 3.13.2 several issues possibly relating to stepper feedback have been encountered. Ghost Y collisions, Ghost XY collisions. Z calibration issues where it requests to calibrate, but reports that it can't, requiring a reset. Y shifting print failures on printing. Machine was fine before 3.13.2 and has had regular maintenance. Upon detecting a collision it tries to recover but often stops to wait for user input to continue. Ghost collisions happen every 5 minutes or so. Stepper motors are noticeably louder, in particular the Y stepper, like a dull drone noise compared to the usual higher pitched stepper motor sound. Motors and board appear to all be within normal thermal levels, with Y motor being very slightly warmer than others. Prusa has been contacted, referred me to GitHub.
  
**To Reproduce**
  Install 3.13.2 build 7080 firmware from 3.11.0 (original firmware shipped with machine) Belt tensions are good at X243, Y273. Measured frame skew is .06 degrees and Y distance from minimum is L10.09mm, R9.88mm Upon downgrading back to 3.11.0 machine operates as before.
  
**Expected behavior**
  Expected normal operation as before, with minor improvements.
  
**G-code**
  Various G-codes that have worked dozens of times exhibit this behavior including new slices done with 2.7.1

**Video**
  Unlikely to assist.

JeremyDWilliams, 2024-02-04:
Update: After many hours/data printing with 3.11.0 machine is still running fine. So it is highly unlikely the machine has an issue. 

The new firmware clearly caused this. But I don't see others commenting here so will try using latest build and report back in case the file was somehow corrupted and this is a one off. 

If anyone has had similar issues after upgrading, I'd love to hear about it. Or positive experiences from those who jumped versions without issues.

rtarb, 2024-02-07:
I am having the same issues noted here. Updated both of my MK3S+ last night and both have the droning sound on the Y motor, and ghost crashes. Both were fine before updating to 3.13.2.

3d-gussner, 2024-03-08:
This should be fixed with https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.3 Please verify and close the issue if solved

3d-gussner, 2024-04-02:
Closing due to no interaction

closed 2024-04-02 by 3d-gussner