prusa3d/Prusa-Firmware #4579: MMU3 PTFE extruder tube tension causing snags on blunted/flat shaped filament (after a cut)

vistax64, 2024-01-26:
**Printer type** - MK3S+
**Printer firmware version** -  LASTEST

**MMU upgrade** - MMU3 (KIT) UPGRADE FROM MMU2S
**MMU upgrade firmware version** - LATEST

**MMU3 PTFE extruder tube tension issue**

What is happening? 

The Bowden PTFE tube used on the MMU3 that goes from the MMU3 selector to the extruder (MK3S+) is by default a specific length that Prusa deemed to work best with their buffer system.  Too short and it will drag and too long, will cause extra filament to go into the buffer increasing filament popping out causing the filament to either retract from the MMU3 idler body or push in too far in the MMU3 unit causing the selector to fail/stall trying to move.

While this is well and good, the issue seems to be…
1)	**Resistance for filament in the tube for pushing filament to the extruder:**  its short enough that when the extruder is at the lowest Z height (starting a print) and moves to the upper left section, there is a lot of tension put on the Bowden PTFE tub, which can cause some resistance. 
       
  see attached
       
       
2)	**When filament has been cut (with the MMU3 cut function), the blunt tip can get stuck in the extruder coupling:**  This is the more serious issue, as when the MMU3 performs a cut as part of its troubleshooting routine, the extra drag on the filament from the shorter PTFE Bowden tube and the angle its put under depending where the print is printing, can cause the blunted tip from the cut to get stuck/snagged on the edges of both the QSM M5 fitting (stock mmu3 hardware) or using PC4-M6 couplings.  This issue does not happen with PC4-M10 couplings*.
  
Here are some more close-up pics showing where the snag occurs on the stock QSM fitting (on the extruder)…
  see attached
       

Here is where the snag occurs on the lip of the printed part on the extruder (where the QSM fits into)…
       
  see attached
       
To solve this, you can use a PC4-M10 coupling in the extruder as the PTFE tube will push all the way down to the PTFE inside the hotend, giving the smoothest possible path for the filament (cut and blunted or normal pointy tip shaped).  However, this causes the PTFE to be lengthened as it has to push further down into the hotend, which increases length of the PTFE tube, which seems to be a hard coded limit in the MMU3 firmware.  While this limit is good to ensure proper filament buffer length to prevent other issues, it does prevent users from using the PC4-M10 mods that were so popular with the MMU2S, which eliminated the snag/drag problems caused by the lips around the coupling and the printed part (see pics).

  see attached
       
[image]
[image]
[image]
[image]
[image]

For Prusa… Is there any way to slightly increase the default Bowden length, at least in my testing giving another 10mm or so in play for the Bowden tube would be sufficient to prevent the kinking of the tube when the extruder is to the leftmost position and the mmu3 is at filament #5 (and the default homed position on the rightmost side of the mmu3).
That would solve issue #1, and second to solve issue #2 would there be a way to redesign the stock parts for the extruder coupling and the mmu3 coupling to incorporate PC4-M10 fittings to allow the smoothest possible path for the filament?
The mmu3 as it is does work fine, its just predicated on the tips always having a pointed shape, either from manually cutting at a 45degree angle or how the tips naturally form in ramming.  These issues I’ve documented here are prevalently only occurring when an error occurs and the MMU3 performs a cut operation, thereby removing the pointed shape tip and leaving a blunted flat shape that is more prone to snagging.  To me at least, this would greatly improve the reliability of the MMU3 beyond the improvements you all already made.  I just don’t see the logic in not using PC4-M10, as it just makes for a much smother path for both pointed shaped tips and blunted (CUT) tips.

Thank you for your time and curious what others may think or if anyone else has noticed this with their MMU3 setups.

3d-gussner, 2024-01-26:
@vistax64 Thanks for the very detailed report and great pictures showing off the issue. Forwarded it to the HW development department.

You can change the length of the bowden length with gcode `M708` see [MMU Registers](https://prusa3d.github.io/Prusa-Firmware-Doc/Prusa-Firmware-MMU/group__register__table.html). 
Register 0x22  Bowden length can be read with `M707 A0x22` and written with `M708 A0x22 X<value in decimal>`.
The `M707` and `M708` are case sensitive so ensure that your serial terminal DOESN'T UPPERCASE this gcode.
In case you use OctoPrint Settings -> Features -> Terminal Auto Uppercase Blacklist add `M707, M708` to prevent the gcode being UPPER-CASED.

I personally like the MINI bowden couplings. Ideally have a modified version of the ptfe-fitting-mini with 4.1 diameter all the way through.
 
https://www.prusa3d.com/en/product/ptfe-fitting-mini
https://github.com/prusa3d/Original-Prusa-MINI/blob/master/DOCUMENTATION/MECHANICAL%20PARTS/mini-fitting.pdf
https://www.prusa3d.com/en/product/extruder-print-head-ptfe-tube-mini
https://github.com/prusa3d/Original-Prusa-MINI/blob/master/DOCUMENTATION/MECHANICAL%20PARTS/mini-fitting-olive.pdf
https://github.com/prusa3d/Original-Prusa-MINI/blob/master/DOCUMENTATION/MECHANICAL%20PARTS/mini-fitting-swivel-nut.pdf

closed 2024-10-15 by github-actions[bot]

Simon-CR, 2025-12-04:
this is still a problem even today.  (not the tube length, but the filament getting stuck at the selector plate entrance