prusa3d/Prusa-Firmware #4580: [QUESTION] Why does XYZ calibration take 5 minutes per corner??

rw3iss, 2024-01-27:
I'd just like an explanation on why we have to sit here and watch this thing spin around the same spot for 5 minutes.... It takes forever to calibrate! Why? 

A) Could it be possible to increase the speed of the calibration movements?

B) Is it necessary to do what seems to be such a redundant "detection" at each corner, spinning on circles for 5 minutes? Could 1 minute of spinning suffice? 💫

3d-gussner, 2024-01-30:
@rw3iss The XYZ calibration depends on so many variables that it takes up to 24 minutes, some printer do it within 10 others need longer.
As this is a procedure that should only run once after
- Factory reset
- big maintenance
- working/changes on the axis parts

we think it is worth the time as it improves the print accuracy by a lot.

The PINDA and its height, the bed traces around the calibration points, the assembly of the printer and more differ and the XYZ cal tries to scan the points no matter which differences there are (of course to a limit).

> A) Could it be possible to increase the speed of the calibration movements?

The PINDA need consistent, repeatable movements to read correctly. Also it behaves differently when moving down or up, so the get valid and repeatable values we need to keep the several parameters the same. We block even other operations during the XYZ calibration to "focus" on it and not add false values to the calibration.
Too fast movements may work on some combination but fail on others (10 to 24 minutes)

> B) Is it necessary to do what seems to be such a redundant "detection" at each corner, spinning in circles for 5 minutes? Could 1 minute of spinning suffice? 😄💫

Again the spin time varies from printer to printer. If the [traces](https://github.com/prusa3d/Heatbed_MK52_magnetic) (within production tolerances) are bit closer to the circles. Here the printer slows down and/or repeats the detection until the different results comparison is within some margin.

> C) Why do so some corners takes ~5 minutes, while others sometimes take ~2 minutes? The 1st, 3rd, and 4th took twice as long as the 2nd.

Same answer as B) some circles are "easier" to detect than others.

Hope that explains a bit why the XYZ calibration takes up to 24 minutes. While it takes "long" it ensures better accuracy in the hundreds of printing hours.

3d-gussner, 2024-02-07:
Closing issue due to no interaction.