prusa3d/Prusa-Firmware #4584: REVO - Thermal Anomaly with 0.8 brass nozzle

LoakAudio, 2024-01-27:
**Printer type** - MK3S+, no enclosure
**Printer firmware version** -  Last REVO Firmware

**MMU upgrade** - No MMU

**SD card or USB/Octoprint**
  SD Card only

**Describe the bug**
  Hi,
 I'm coming to tell you about a problem (not too serious) with the REVO firmware and Thermal calibration.
 I use the last REVO firmware and except the hotend my MK3S+ is stock.
 I use brass nozzles. With 0.25, 0.4 and 0.6 nozzles everything is fine.
With 0.8 nozzle I have many thermal anomalies on layers 2, 3 and 4, never on the first layer, never after the 4th layer. I'd say it goes with the acceleration of the nozzle fan (amateur deduction). I print Prusament PLA, nozzle and heatbed temps are very stable. Thermal Anomaly makes the printer beep but doesn't stop the print and all my prints have finished correctly.

It seems to me that someone reported something similar with 0.8 brass nozzle but I can't find it again, so , sorry if it is a duplicate I will delete if you want.

Open for all tests you could ask.

  
**To Reproduce**
 Try to print something with a big surface on the bed, as I use 0.8 nozzle for big pieces.
  
**Expected behavior**
  No thermal anomalies on the first layers.

3d-gussner, 2024-01-28:
@LoakAudio With changing any component that is related to the thermal behavior of the nozzle you may need to calibrate or adjust the TM settings.

On the first layer normally the part cooling fan is disabled but it turns on at the 2nd layer as speeds are going up.
We also saw some TM warnings on very specific prints/geometries where the part cooling air was blown back direction nozzle and so cooled down the nozzle more than normally expected by the TM.

As the 0.25, 0.4 and 0.6 are working well I would assume that you ran the TM model with 0.4mm. The default TM M310 W tolerance compensates the thermal changes better for both close to 0.4 nozzles at all heights but is outside for the 0.8 at the 2nd to 4th layer.

If you want to analyze this you need to log the serial output while lowering the Warning value to see the differences between the different nozzle sizes.

`M310 W0.2` instead of `M310 W0.85` (default) to trigger warnings earlier or in this case to "enable" logging of the thermal model.
I would also temporary disable the Beeper while collecting data with `M310 B0`.

With `M310` you get the current TM values. 
You can also run the TM calibration for the 0.8 nozzle and a modify your profiles to set in start gcode the new TM values and reset them to the default TM values.

Hope that helps a bit to find a solution. Please also checkout E3D support / discord as this is a 3rd party product.

3d-gussner, 2024-03-08:
Closing due to no reply or feedback

closed 2024-03-08 by 3d-gussner