prusa3d/Prusa-Firmware #4585: Problem with i3 MK2S firmware...

DeWisz, 2024-01-28:
Hi everybody!
I have i3 MK2S Prusa and decided to add a filament runout sensor.
I was trying to use directions from here: https://www.thingiverse.com/thing:5136321
I copied all steps directly, but ended up with unresponsive printer with some lit squares on the printer display...
It seams that my computers skills are way below of what is needed to complete this task.
Has anybody a .hex file with firmware that was build in same exact way so I could flash my printer and just wire up my sensor?
Or, is there anybody who could build such firmware for me as stated in link above?

3d-gussner, 2024-01-29:
@DeWisz Nice to see a MK2S question and that it is still used after all these years.

I have an very old MK2S firmware that should work for you. Please checkout https://github.com/3d-gussner/Prusa-Firmware/releases/tag/v3.1.0-001-HFM
This isn't the latest firmware for the MK2S but works/ed well on my MK2S.
Checkout https://www.printables.com/search/models?q=mk2s%20filament%20runout%20sensor and especially https://www.printables.com/model/346306-prusa-mk2s-dust-filteropto-filament-runout-sensor

TheZeroBeast contributed to the MK2S firmware we made in the past and his sensor has been tested and is working well.

As far I can see your Sensor should also work with this firmware.

> I copied all steps directly, but ended up with unresponsive printer with some lit squares on the printer display...
> It seams that my computers skills are way below of what is needed to complete this task.
> You need to select the I copied all steps directly, but ended up with unresponsive printer with some lit squares on the printer display...
> It seams that my computers skills are way below of what is needed to complete this task.

Did you select the [MK2 branch](https://github.com/prusa3d/Prusa-Firmware/tree/MK2 ) as your base to compile the firmware?
Compiling the firmware back than wasn't easy see https://github.com/3d-gussner/Prusa-Firmware/blob/v3.1.0-001-HFM/BUILDING.md
You NEED to use [ARDUINO IDE 1.8.5 ](https://github.com/arduino/Arduino/releases/tag/1.8.5) and I would advise to set it up in [portable mode](https://docs.arduino.cc/software/ide-v1/tutorials/PortableIDE/)  to prevent issues with other Arduino IDE versions, also disable the update function to stay at know stable 1.8.5 build environment for the MK2 branch build.

Hope that helps.

DeWisz, 2024-01-30:
@3d-gussner  
Hi there!
Thanks a lot for your respose. 
I flashed firmware with a hex file from first link you've provided, connected a sensor and tried out to print something. 
Sensor didn't work - mostly because I didn't turn it on via LCD Settings :-). Unfortunatelly, before I've realized that, I've flushed the firmware again. From this moment on, every time I try to print something, I'm getting a memo to update to the 3.2.3 version...
Ps. Sorry for late response, but I had to try all options and make sure not missing any step down the road...

3d-gussner, 2024-01-30:
I have no idea why your comment isn't shown here.
_Hi there!
Thanks a lot for your respose.
I flashed firmware with a hex file from first link you've provided, connected a sensor and tried out to print something.
Sensor didn't work - mostly because I didn't turn it on via LCD Settings :-). Unfortunatelly, before I've realized that, I've flushed the firmware again. From this moment on, every time I try to print something, I'm getting a memo to update to the 3.2.3 version...
Ps. Sorry for late response, but I had to try all options and make sure not missing any step down the road..._

> I'm getting a memo to update to the 3.2.3 version...

That is correct as the firmware is based on 3.1.0 and you PrusaSlcier Profile and gcode files contain a notification to upgrade the firmware to 3.2.3. You can ignore it.

Please verify first if your FSensor works correctly.

I am not sure when and if I will have time to update my old firmware to 3.2.3 as not many ask for that.

DeWisz, 2024-01-31:
Hi again!
Everything is working fine now. You were right; it was a mechanical problem with the sensor.
This is my first Prusa, and I must admit, I'm very impressed by its reliability - disregarding the age of my printer.
I got hooked up, and my next printer will definitely be a Prusa too.

Thanks again for your support, and see you soon!

3d-gussner, 2024-02-01:
@DeWisz Glad to hear that you enjoy your "ancient" :grinning: but reliable Original Prusa i3 MK2. :heart_eyes: 
Now it is a MK2.1 as you added the Filament Runout Sensor.

The Prusa MK2 was also my first printer back in 2016 and I still like it and will NEVER upgrade to MK2.5/S as I like the MK42 bed without the steel sheet for some prints.

Please consider to close this issue as it is solved/answered your question. I am trying to keep the issue tacker clean and small.

P.s.: If you need me to create a newer 3.2.3 version with the v3.1.0-001-HFM changes, please open a ticket [there](https://github.com/3d-gussner/Prusa-Firmware/issues) and I will try to get this done in some spare time.