prusa3d/Prusa-Firmware #4588: [BUG] printer tries to unload filament with cooled down nozzle (MMU3)

slaesh, 2024-02-03:
**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
MK3S+ + Revo Nozzle

**Printer firmware version** -  [e.g. 3.8.1, 3.8.1-RC1, ...]
3.13.2 Revo

**MMU upgrade** - [e.g. MMU2S, MMU2, MMU1]
MMU3

**MMU upgrade firmware version** - [e.g. 1.0.6, 1.0.6-RC2, ...]
MMU3_3.0.1+837.hex

**SD card or USB/Octoprint**
PrusaLink

**Describe the bug**
Had a print this night, first layer wasnt ideal, but did not look that bad. I let him continue in the hope of "recovery", dont need it beauty. :)
Bad luck, it did not succeed. This morning is saw "crash detected", printer asked me "continue or stop". I chose stop, since it looked pretty bad..
Bed was heated, but the nozzle was "cold"..
Then it wanted to unload the filament (I didnt noticed the cold nozzle yet) and hit yes.. You can imagine what happened, nothing! loud sound, gears melting the stuck filament ^^

  
**To Reproduce**
Hard
  
**Expected behavior**
If user hits yes, unload, .. check if the nozzle is hot though! ;) if not, heat :))

gudnimg, 2024-02-03:
Hi, A fix was merged recently: https://github.com/prusa3d/Prusa-Firmware/pull/4582

It will be included in 3.14.0 release. No ETA though.

closed 2024-02-03 by slaesh

chrisboer, 2024-02-20:
Have the same issue

3d-gussner, 2024-02-20:
@chrisboer Thanks for reporting but this is already fixed see https://github.com/prusa3d/Prusa-Firmware/issues/4588#issuecomment-1925301862

chrisboer, 2024-02-20:
I know when will v3.14.0 become available ?

3d-gussner, 2024-02-20:
@chrisboer We are still working on it can't say when it will be released. At the moment we try to get FW3.13.3 asap tested and released. See https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2 Attention part from yesterday