prusa3d/Prusa-Firmware #4595: [ENHANCEMENT]  Revo Firmware:  Do Not Heat Extruder When Initiating Nozzle Change

samsmaxwell, 2024-02-12:
**Printer type** - MK3S/+ with REVO Hotend
**MMU Upgrade** - N/A

**Is your enhancement related to a problem? Please describe.**
The MK3S/+ firmware includes an option under Settings/HW Setup/Nozzle Change that centers the extruder on the X axis, raises it to the maximum height, and heats the nozzle to 280C.  Revo hotends are designed to be swapped cold and with bare fingers, so heating the nozzle prior to a nozzle change can actually result in injury to the user.

  
**Describe the solution you'd like**
The requested enhancement is to modify the Revo firmware behavior to set the hotend temperature to 0 during a nozzle change, rather than setting the extruder temperature to 280C.

3d-gussner, 2024-02-12:
@magistersmax Thanks for the report and makes sense to change it. I think easiest would be to hide the Nozzle Change completely for REVO users.
Is that an option?

magistersmax, 2024-02-12:
I guess it’s one way to do it, it’d certainly be the easier way to go.  

It is convenient to have the option to raise and center the extruder without manually moving axes, so I think it would be better to keep the option and modify its behavior.  If you keep the menu option but set temp to 0, you would want to change the prompt text too.  It shouldn’t say that the nozzle is at 280C, and a short warning to ensure the nozzle is cool enough to touch might be good.

closed 2024-04-11 by 3d-gussner