prusa3d/Prusa-Firmware #4596: [BUG] MK3SMMU3 creates some bad layers

gleibfried, 2024-02-13:
**Printer type** - MK3S+
**Printer firmware version** -  3.13.2-7080

**MMU upgrade** - MMU3
**MMU upgrade firmware version** - 3.0.1

**SD card or USB/Octoprint** - SD card

**Describe the bug**
My printer creates some bad layers, like as if the layer was a little too crushed (like when the Live Z is too low). It happens randomly, like it's going well for 2 centimeters, then one bad layer, then 2 milimeters fine, then 1 bad layer, etc etc.

You can see it on this picture: https://drive.google.com/file/d/1RWJlnD3WsorDRPME7E2RO0fqd4lpA_gw/view?usp=drive_link

It's happening on a whole layer around the whole body, so it's not like a tiny layer shift.
I print it using some regular printing profiles, similar to some of my other gcodes which have had no issues.
It's something I already printed before with no issue on my MMU2.

I've contacted the support which directed me here as there seemed to be no issues on my end (the belt tension is fine too).
  
**To Reproduce**
Mainly printing the same gcode as I.
  
**Expected behavior**
Similar bad layers happening?
  
**G-code**
https://drive.google.com/file/d/1W79tOz2SFTu13cMhiydFz7t4qK2Ufik-/view?usp=drive_link

3d-gussner, 2024-03-08:
@gleibfried Please update to https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.3 and let us know if that solves your issue.

gleibfried, 2024-03-08:
I have updated and made a couple of checks on the printer and the issue is indeed solved! Thank you

3d-gussner, 2024-03-08:
@gleibfried Thanks for the feedback and closing the issue.