prusa3d/Prusa-Firmware #4597: [BUG] Error 17604 on Prusa XL 2T during power-on

Marinosbitter, 2024-02-15:
**Printer type** - Original Prusa XL (2 toolhead)
**Printer firmware version** -  5.1.2+13478

**SD card or USB/Octoprint**
SD Card

**Describe the bug**
During power-on it gives an error: 
Error! #17604
Firmware file missing in the USB flash!

Clicking on the rotary knob continues startup and printer functions as normal.
  
## Symptoms:

### When the error triggers:
* Power on with an empty USB gives the error
* Power on with an USB with gcode files gives the error
* Power on with an USB with firmware file gives the error
* Power on with USB inserted gives the error, but clicking the rotary knob continues the startup and from then on, the printer works * completely normal

### No error get triggered:
* Power on without USB inserted gives no error and starts normal
* Connecting a USB after startup works as normal
* Rest of the functionality of the printer works as normal, both after startup with the error as after startup without the error
* I can print form the USB
* I can flash the firmware using USB
* I can print from Prusa Connect
* I can upload files trough Prusa Connect
* I can flash the firmware using Prusa Connect
* Printer self tests are all OK

## Solutions tried:
* Different USB (both USBs work with my Prusa MINI as normal)
  * Tried original USB that came with the printer
  * Tried a SanDisk 32GB USB3.0
* Formatting (quick) both USBs
* Formatting (complete) both USBs
* Flashing to older firmware
* Flashing to same firmware
* Flashing to new firmware trough Prusa Connect
* Reconnecting the grey ribbon cable behind the LCD
* Reconnecting the orange ribbon cable behind the LCD

JonMrowczynski, 2024-02-16:
Hello Marinosbotter, I hope that you have been doing well!

It appears that your issue is for the XL. The repository for this can be found [here](https://github.com/prusa3d/Prusa-Firmware-Buddy). As stated in the README for [this repository](https://github.com/prusa3d/Prusa-Firmware?tab=readme-ov-file#prusa-firmware-mk3) "This repository contains the source code and the development versions of the firmware running on the [Original Prusa i3](https://prusa3d.com/) MK3S/MK3/MK2.5S/MK2.5 line of printers."

From my understanding of what is usually done in this case, it would be most helpful for you to close this issue here and repost it on the [XL firmware repository's issue tracker](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues) so that it gets seen by the right people.

Marinosbitter, 2024-02-16:
Hi Jon,

Thanks for the heads-up. Apparantly I should not be posting bugs late at night. I've added the issue here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3762 and I'll close this one.

closed 2024-02-16 by Marinosbitter