prusa3d/Prusa-Firmware #4604: [Community made] MK2.xS printer community build and support

3d-gussner, 2024-02-20:
Prusa Research will NOT follow up these issues!
The maintainers of the "Community made" feature should/will react.

**Which Community made feature do you want to address?**
I have seen few users working on getting the MK2S printers working with the current firmware.

I would suggest to define few things:
Printers with own variants:
MK2.0S = commenting out the FSensor is quiet challenge and I would drop this one as the FSensor can be disabled in MK2.1S 
MK2.1S = MK2S + IR sensor connected
MK2.2S = MK2S + IR sensor + Hotend fan tacho
MK2.3S = MK2S + IR Sensor + Hotend fan tacho + SUPERPINDA
MK2.4S = MK2S + IR Sensor + Hotend fan tacho + SUPERPINDA + BondTech gears and extruder body

No MK2.x as the PAT9125 isn't available anymore.
Difference to MK2.5S would be the MK42 bed vs. MK52 bed and Sheets

As the MK2/S are EOL this would be a community driven firmware and the owners would be responsible to test, maintain and provide support for these.

**What is your request/question/suggestion?**
Anyone interested? Please leave a :+1:  or :-1:  to vote for it.

@GilesBathgate @matz9x @vsky279 I have seen that you were working on a custom firmware for your own printers, would you be interested to work on this?

vsky279, 2024-02-20:
Yes indeed. I have recently upgraded my old MK2 to MK2+magnetic bed.
I manage to make it run the 3.13.2 firmware with disabled fan check and filament sensor off. I call it MK2.4.
Otherwise I'm a bit lost in all your variants... :-)

3d-gussner, 2024-02-21:
@vsky279 Thanks for the answer. So you have a MK2 with a MK52 12V bed and a steel sheets, or do you have a different solution for the magnetic bed?

vsky279, 2024-02-21:
That's correct. MK2 + MK52 12V bed and a steel sheets.

3d-gussner, 2024-02-21:
The idea is to make it possible to compile the current firmware for different types of MK2S with partial upgrades of MK2S -> MK2.5S
This upgrade contains 5 changes
1. IR sensor to detect filament runouts
2. Hotend fan with tacho to prevent heat creeps with faulty fan
3. SuperPINDA which doesn't need the PINDA temp calibration
4. Bondtech dual gears and different extruder body
5. MK52 12V bed with SteelSheet.

So making it possible to en/disable each of these upgrades enables the community to use the MK2S with any combination of the upgrade parts above.

Somebody may want to have IR Sensor, SuperPINDA and MK52 12V bed and would be able to create a new variant to do that.

I personally will NEVER change the MK42 bed on my MK2S+IR sensor as it is perfectly good and even for some prints better as there is no sheet that may be pulled up at the corners when filament cools down to fast.

Hobo2k, 2024-02-28:
> Yes indeed. I have recently upgraded my old MK2 to MK2+magnetic bed. I manage to make it run the 3.13.2 firmware with disabled fan check and filament sensor off. I call it MK2.4. Otherwise I'm a bit lost in all your variants... :-)

I had this setup also but get's alwas a leveling error.
Can you send me your FW? It would be verry nice.

3d-gussner, 2024-02-28:
@Hobo2k I don't have a firmware for the MK2.0-MK2.4. This is ticket to look for volunteers and see if there is any demand for it.

Hobo2k, 2024-02-28:
Yeah, a demand for MK2S with the MK52 bed 😉 

But I thought, that @vsky279 had a FW for this setup, if I understand his post correctly 😃

vsky279, 2024-02-28:
@Hobo2k Felix, you can try to use one of these compiled firmwares.
[https://github.com/vsky279/Prusa-Firmware/tree/MK24-3.13.2-binary/PF-build-hex/FW3132-Build/BOARD_RAMBO_MINI_1_3](https://github.com/vsky279/Prusa-Firmware/tree/882291e447b0a296d5675db922aa4ec5c247f67d/PF-build-hex/FW3%EF%80%8D13%EF%80%8D2%EF%80%8D-Build/BOARD_RAMBO_MINI_1_3%EF%80%8D)
Be careful.

Better approach would to compile your own as probably you don't need some features that are implemented in this fw. i.e.
- it seems that my calibration point no. 3 is somehow bad so I have modified the XYZ calibration process so that only points no. 1, 2 and 4 are used
- serial output is redirected to UART2 (I'm using ESP3D feature)

You can clone your own branch and remove these commits if needed.

sl1pkn07, 2024-05-11:
this!

i have an mk2s in process to convert in mk2.5s+ and plus disable mmuX for use the free serial port as esp3d serial port

thnx @vsky279 i take look you firmware about support esp3d and the experimental Flashair branch (also i have a unit for all my printers out of prusa)

@3d-gussner what is the " PAT9125" thing?

greetings

sl1pkn07, 2024-07-06:
Question

MK2.5S+ with Einsy board is still considered as MK2.x?

i've switch today to this configuration (powerpanic include, all in 12v excepts motors and board iself). but i got fail calibration Z or XYZ in Z step.  firmware is a bit messy (hardcoded values in random code) for figure out what happen

any throughts?

greetings

Tirpitz93, 2024-07-09:
I will be trying to port as many features as I can to my setup.
 - MK2.5S @ 12V (RAMBo13a board)
   - E3D REVO hotend (40w 12V)
 - MMU2S @ 24V

sl1pkn07, 2024-11-19:
Hi. question is possible bring back support to MMU1 (yes, 1) to mk2.5/+ variant?

greetings

3d-gussner, 2024-11-20:
@sl1pkn07 I don't think that will be easy task as there have been some purge done to get more free resources for other features.
If someone from the community is willing to get the MMU1 working we will consider to add it.

3d-gussner, 2024-11-20:
>MK2.5S+ with Einsy board is still considered as MK2.x?

sl1pkn07, 2024-11-20:
tnx

3d-gussner, 2024-11-20:
@sl1pkn07 What is your setup? Einsy + MK42 12V bed + MMU1 ?

sl1pkn07, 2024-11-20:
no. two different machines

MK2.5s+ with einsy (+ MMU2S):
MK52 12v hotbed variant and 12v hotend
24v for motors and electronics powered by 3A step-up booster connected directly to PSU.
clone (chinese) powerpanic circuit (works perfect with 12v) also conected to PSU

MK2S MMU1:
i want convert to MK2.5s+ like above, but with the MK3S+ extruder modificated for integrate the MMU1 Y-splitter. and use the MMU1 multiplexer circuit