prusa3d/Prusa-Firmware #4605: Firmware 3.13.2,  PrusaSlicer 2.7.2 alpha2, Does not detect M84 at end of gcode file from SD card

foxkit, 2024-02-20:
**Printer type** - MK3S
**Printer firmware version** -  3.13.2

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 3.0.1

**SD card or USB/Octoprint**
  SD card

**Describe the bug**
  When printing a file from the SD card, I get an error that the file is incomplete.  Reading bug reports, that is an indication that the M84 is missing. 

I suspect that the M84 is scanned in the last few blocks of the file, and the configuration dump has grown to be longer than the look-back of the M84 scan.
  
**To Reproduce**
I don't know why the file is as it is.  The "M84" is found on line 256783 (character 5,901,286) of the G-code file.  When I moved the M84 to the end of the file (line 257,111, character ~5,923,060), the file is not reported to be in error.
  
**Expected behavior**
 I expect the file to print without a warning.
  
**G-code**
 [Einstein-r2_0.2mm_PLA,PLA_MK3SMMU2S_6h0m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/14351788/Einstein-r2_0.2mm_PLA.PLA_MK3SMMU2S_6h0m.gcode.zip)

**Video**

3d-gussner, 2024-02-21:
@foxkit Thanks for reporting and very good description.
We have fixed that in https://github.com/prusa3d/Prusa-Firmware/pull/4507 and will be in FW3.14.0
I can't say when FW 3.14.0 will be released, we are testing the latest build.

3d-gussner, 2024-02-21:
Please consider to close this issue as it has been dixed.

foxkit, 2024-02-22:
I'm sorry this was a duplicate of a known issue.  I didn't search for the right terms.  It would have been better if I had checked the source code.

I consider it to be fixed and closed. 

Thank you.

closed 2024-02-22 by foxkit

3d-gussner, 2024-02-23:
@foxkit Don't worry, searching on github is okay but you may need not find things that easy. Thanks for closing the issue keeps the issue tracker bit cleaner.