prusa3d/Prusa-Firmware #4610: [BUG] Changing end-code has no effect after print finished

Acidham, 2024-02-24:
**Printer type** - MK3.5
**Printer firmware version** -  5.2.0+13881

**SD card or USB/Octoprint**
  USB and PrusaConnect

**Describe the bug**
 Changing End-Code has no effect
  
**To Reproduce**
  I tried to change Y value of following line `G1 X241 Y204 F3600` to prevent heatbed to hide the new XLCD. I changed to Y181. After finishing the print the heatbed still stops at Y204
  
**Expected behavior**
 Heatbed arrives at Y181 to unhide the LCD
  
**G-code**
 G1 X241 Y181 F3600

**Video**
 
[image]d

gudnimg, 2024-02-24:
Hi from your image I see its an MK4, this is not the correct repo to report that issue. This repository is only for the 8-bit firmware printers (MK3S+ and older machines)

The repository for MK4 (as well as the other 32-bit firmware printers) is found here https://github.com/prusa3d/Prusa-Firmware-Buddy/issues

Acidham, 2024-02-24:
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3790

Acidham, 2024-02-24:
> Hi from your image I see its an MK4

Mk 3.5 ;)

closed 2024-02-24 by Acidham