prusa3d/Prusa-Firmware #4615: [Community made] FIXED MiniTemp Firmware

Ethan-Ming, 2024-02-25:
Prusa Research will NOT follow up these issues!
The maintainers of the "Community made" feature should/will react.

Please, before you create a new "Community made" ticket, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Which Community made feature do you want to address?**
MinTemp Error in winter
**What is your request/question/suggestion?**
Screw this feature 

** This Firmware Disable MinTemp protection because I'm tired waked by the beeping sound in the winter morning

[Screw MINTEMP for all printers(MK2,MK3,MINI).zip](https://github.com/prusa3d/Prusa-Firmware/files/14395087/Screw.MINTEMP.for.all.printers.MK2.MK3.MINI.zip)

3d-gussner, 2024-02-26:
@Ethan-Ming Hi, I removed the link to the zip file. Please don't understand that as censuring and read my comment below.

Please do not share ZIP files only without any reference to the source code. I couldn't find any fork on your github and so nobody can review your changes. If you want to provide a Community made version, release it on your gihub and feel free to share the link to your fork.

I release myself in the past few MK2 related community builds and users could see all changes that have been made and download the firmware on their printers. See as an example https://github.com/3d-gussner/Prusa-Firmware/releases/tag/v3.1.0-001-HFM

This is open source and you also should provide the source code and not only hex files. And as open source anyone can modify the printers to their needs.

3d-gussner, 2024-03-13:
Closing issue as no interaction

closed 2024-03-13 by 3d-gussner