prusa3d/Prusa-Firmware #4616: [BUG] First layer is off when a print is started after a cancelled one on 3.13.2 revo MK3s+

adrienafl, 2024-02-25:
**Printer type** - MK3S
**Printer firmware version** -  3.13.2

**SD card or USB/Octoprint**
Octoprint 

**Describe the bug**
When starting a print just after cancelling one (from the printer), the printer prints too high which makes the first layer wrong. Mesh bed leveling looks good, the print head doesn't seem too high during this phase. This issue is gone after restarting the printer but comes back right after cancelling a print
  
**To Reproduce**
- Start a print from octopi
- Cancel it from the printer
- restart it from octopi
  
**Expected behavior**
First layer should be right
  
**Video**
  Please attach a video. It usually helps to solve the problem.

adrienafl, 2024-02-26:
Rolled back to 3.13.1, issue is gone

3d-gussner, 2024-03-08:
@adrienafl Just verified that it is working with 3.13.3 and previous versions.
Do you have in OctoPrint -> Settings -> GCODE Scripts
- After print job is cancelled `M603`
- After print job is paused `M601`
- Before print job is resumed `M602`

adrienafl, 2024-03-08:
Only thing I have is : After print job is cancelled 

```
; disable motors
M84

;disable all heaters
{% snippet 'disable_hotends' %}
{% snippet 'disable_bed' %}
;disable fan
M106 S0
```

How would I only have the issue with 3.13.3 and not 3.13.1 ?

eltron247, 2024-04-10:
I just observed this same behavior in 3.13.3 on my MK3S.
First layer calibration looks good and runs correctly. As soon as I go to print It commands the nozzle high. The LCD said it was commanding a height of 0.4 instead of 0.2.

I tested with the print twice before resetting and it happened consistently.
Hard resetting the printer via the button allows this to run correctly again.

adrienafl, 2024-04-10:
I rolled back to 3.13.1 and issue is gone. I'll upgrade to MK3.5 soon so I didn't want to investigate furtger

closed 2024-06-17 by github-actions[bot]