prusa3d/Prusa-Firmware #4621: [BUG]<Enter comprehensive title>Alpha 6.0.0 Firmware Bug Tool head 3  Dock Crash

bluerider098, 2024-02-29:
-->
**Printer type** - Prusa XL 5T
**Printer firmware version** -  6.0.0 Alpha

**SD card or USB/Octoprint**
  USB

**Describe the bug**
  Preheated Printer for PETG.  Manually selected toolhead 3 and used control move axis to move the y axis close to the front of the machine.  Raised nozzle temp to 230 and brushed off nozzle.  Manually lowered temp for nozzle 3 back to 170.
Selected print from the printer and selected file.
 Printer for normal startup routine with heat soak and then probed the bed.  After bef probing completed the print head shot back towards dock 3 and violently crashed into it causing the dock crash error to come up on the screen.

At this point I didn't want to proceed anymore so I got the reset button and reverted back the firmware to the currently released version. 
  
**To Reproduce**
  Please describe steps to reproduce the behavior.
  
**Expected behavior**
  A clear and concise description of what you expected to happen.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

JonMrowczynski, 2024-02-29:
Hello bluerider098, I hope that you have been doing well!

It appears that your issue is for the XL. The repository for this can be found [here](https://github.com/prusa3d/Prusa-Firmware-Buddy). As stated in the README for [this repository](https://github.com/prusa3d/Prusa-Firmware?tab=readme-ov-file#prusa-firmware-mk3) "This repository contains the source code and the development versions of the firmware running on the [Original Prusa i3](https://prusa3d.com/) MK3S/MK3/MK2.5S/MK2.5 line of printers."

From my understanding of what is usually done in this case, it would be most helpful for you to close this issue here and repost it on the [XL firmware repository's issue tracker](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues) so that it gets seen by the right people.

bluerider098, 2024-02-29:
Ok no problem. Sorry about that

 I posted here because that is where the link on the Prusa forum sent me.  Maybe we need to get with them and tell them to post the correct link.  See attached.

Thanks,

Brian Killian

bluerider098, 2024-02-29:
Moved to XL section.

closed 2024-02-29 by bluerider098