prusa3d/Prusa-Firmware #4629: [FEATURE REQUEST] SNMP support

GRIFFCOMMca, 2024-03-08:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

SNMP support for the base printer and possible (later date) the MMU units (i would think the Mk3, Mk4, XL and Mini would be a great start.

Although this is not related to a problem, many larger companies use ZABBIX which will read SNMP managment data and create alerts and graphs to show history and current status of the hardware

As many usable SNMP OIDs as possible, we are willing to help create a database of usable OIDs and how that data would be read and why.  We have created a number of templates to read SNMP data from Grandstream products (desk phones, DECT phones and the IP PBX).

Many Thanks

sarusani, 2024-03-10:
I might be wrong here, but isn't SNMP based on UDP?
If so, you want to move this feature request to the [Firmware-Buddy](https://github.com/prusa3d/Prusa-Firmware-Buddy) repo. This firmware (for MK3S/MK3/MK2.5S/MK2.5) does not offer network communication and therefore can't implement SNMP.

GRIFFCOMMca, 2024-03-10:
> I might be wrong here, but isn't SNMP based on UDP? If so, you want to move this feature request to the [Firmware-Buddy](https://github.com/prusa3d/Prusa-Firmware-Buddy) repo. This firmware (for MK3S/MK3/MK2.5S/MK2.5) does not offer network communication and therefore can't implement SNMP.

Yes, port 161 UDP, and yes this would only be aimed at the boards that have network built in (Mk4, MINI, XL), i thought there was only a single forum for all firmwares.... will re-create this over at the link you have above.

GRIFFCOMMca, 2024-03-10:
Closed here

GRIFFCOMMca, 2024-03-10:
> I might be wrong here, but isn't SNMP based on UDP? If so, you want to move this feature request to the [Firmware-Buddy](https://github.com/prusa3d/Prusa-Firmware-Buddy) repo. This firmware (for MK3S/MK3/MK2.5S/MK2.5) does not offer network communication and therefore can't implement SNMP.

Yes, port 161 UDP, and yes this would only be aimed at the boards that have network built in (Mk4, MINI, XL), i thought there was only a single forum for all firmwares.... will re-create this over at the link you have above.  Have closed this

closed 2024-03-10 by GRIFFCOMMca