prusa3d/Prusa-Firmware #4633: Can we have Time Remaining in the Footer?

ClutchplateDude, 2024-03-12:
I use OctoPrint to print and it was great with the MKS3+ because the 20x4 LED would show the very accurate remaining time (OctoPrint's calculation is bad). After upgrading to MKS3.5, I no longer have remaining time displayed on the printer screen.

**Describe the solution you'd like**
One of the Footer options should be Remaining Time

**Describe how it would work**
You would select it in the Footer menu and during a print it would show the time remaining. At other times, it could just show 0:00

**Describe alternatives you've considered**
Another alternative would be to display Time Remaining somewhere else on the OctoPrint screen (that octopus logo doesn't need to take up half the screen like that :-) )

**Additional context**

3d-gussner, 2024-03-12:
@ClutchplateDude This is the MK2/S, MK2.5/S and MK3/S/+ repository. Please reopen your request here https://github.com/prusa3d/Prusa-Firmware-Buddy as this is for the 32-bit printers MINI, MK3.5, MK3.9, MK4, XL, ... repository

Also close this issue here. Thanks

ClutchplateDude, 2024-03-12:
Wrong repo for this request.....

closed 2024-03-12 by ClutchplateDude