prusa3d/Prusa-Firmware #4634: [QUESTION] Add 3 more material slots to MMU2S for MK3S

Axolotelking, 2024-03-12:
Hallo I added 3 more material slots for the MMU2S for the MK3S.

I would like to know what I need to change in the code to add T5-T7 in the G-Code.

I already added the T5-T7 function on the MMU2S firmware and tested them by swaping the T1-T3 with them.

When useing Powershell I can monitor the print and I get no response if there are T5-T7s in the G-Code.

I use the Prusa-Firmware-3.12.2 and the MM-control-01-1.0.6.

I already changed the the Extruder value in the MMU code from 5-8.

In the Prusa-Firmware-3.12.2 I changed the values in Marlin_main.cpp from 4-7 and added  T5, T6 and T7 in mmu.h in the line 
enum class MmuCmd : uint_least8_t.

else if(code_seen('T'))
  {
      
      int index;
      bool load_to_nozzle = false;
      for (index = 1; *(strchr_pointer + index) == ' ' || *(strchr_pointer + index) == '\t'; index++);

      *(strchr_pointer + index) = tolower(*(strchr_pointer + index));

      if ((*(strchr_pointer + index) < '0' || *(strchr_pointer + index) > '7') && *(strchr_pointer + index) != '?' && *(strchr_pointer + index) != 'x' && *(strchr_pointer + index) != 'c') {
          SERIAL_ECHOLNPGM("Invalid T code.");

Dose some one know what else I have to change, to be able to use T5-T7 in the G-code ?

Thank you very much for your time!
Bernd

3d-gussner, 2024-03-12:
@Axolotelking Hi, you have to use newer versions of the 
- Prusa-Firmware https://github.com/prusa3d/Prusa-Firmware/tree/v3.13.3
- MMU firmware https://github.com/prusa3d/Prusa-Firmware-MMU/tree/v3.0.2

In MMU firmware you need to modify the `/src/config/config.h` to your design 
- [toolCount](https://github.com/prusa3d/Prusa-Firmware-MMU/blob/v3.0.2/src/config/config.h#L24)
- Selector see https://github.com/prusa3d/Prusa-Firmware-MMU/blob/v3.0.2/src/config/config.h#L155-L179
- Idler see https://github.com/prusa3d/Prusa-Firmware-MMU/blob/v3.0.2/src/config/config.h#L155-L179

In the Prusa-Firmware you gonna need to modify
- `/Firmware/variants/MK3S.h`
  - [MMU_FILAMENT_COUNT](https://github.com/prusa3d/Prusa-Firmware/blob/v3.13.3/Firmware/variants/MK3S.h#L651)
- `/Firmware/ultralcd.cpp`
  - [load_all_wrapper](https://github.com/prusa3d/Prusa-Firmware/blob/v3.13.3/Firmware/ultralcd.cpp#L4773) replace `i < 5` to `i < MMU_FILAMENT_COUNT`
  - [loading_test_all_wrapper](https://github.com/prusa3d/Prusa-Firmware/blob/v3.13.3/Firmware/ultralcd.cpp#L4850) replace `i < 5` to `i < MMU_FILAMENT_COUNT`
   - [loading_test_all_wrapper](https://github.com/prusa3d/Prusa-Firmware/blob/v3.13.3/Firmware/ultralcd.cpp#L4850) replace `i < 5` to `i < MMU_FILAMENT_COUNT`
-  `/Firmware/Tcodes.cpp`
  - [IsInvalidTCode](https://github.com/prusa3d/Prusa-Firmware/blob/v3.13.3/Firmware/Tcodes.cpp#L15) `s[i] > '4'` to `s[i] > '7'`

I may have missed few things but I hope you can continue.

May I ask for a picture of your design? Always nice to see the community modifying the printers and MMU.

Axolotelking, 2024-03-12:
Hi I alredy changed a lot in the old software is it not possible in the old version?

I will send a picture when i finished the testing and assemble  it compleatly.

3d-gussner, 2024-03-12:
> Hi I already changed a lot in the old software is it not possible in the old version?

Please don't waste your time on the old firmware as the MMU development is now on Prusa-Firmware-MMU.

Axolotelking, 2024-03-13:
Hi I used the new firmware and changed the lines you told me about, but during the test today the print imediently stopped whenever T5, T6 or T7 were in the G-Code.

Is this error only in my code or do I have to change more?

3d-gussner, 2024-03-14:
> Hi I used the new firmware and changed the lines you told me about, but during the test today the print imediently stopped whenever T5, T6 or T7 were in the G-Code.
> 
> Is this error only in my code or do I have to change more?

As I wrote above these changes I have found may not be enough and you need to look deeper. I have found these changes within few minutes so I could point you in the correct direction.

3d-gussner, 2024-03-14:
A suggestion: Create a fork of both firmwares and publish your changes there, the community may be able to help you.
@ComunidadMaker3D has been working on a 10 slot MMU and maybe you find in these forks of [Prusa-Firmware](https://github.com/ComunidadMaker3D/CM3D-Prusa-Firmware) and [Prusa-Firmware-MMU](https://github.com/ComunidadMaker3D/CM3D-Firmware-MMU) something that helps.

Axolotelking, 2024-03-14:
Ok thank you very much!

3d-gussner, 2024-03-18:
@Axolotelking Please consider to close this issue to keep the issue tracker small.

closed 2024-03-19 by Axolotelking