prusa3d/Prusa-Firmware #4637: [BUG]<Enter comprehensive title>

chrisboer, 2024-03-15:
**Printer type** - MK3S 
**Printer firmware version** -  3.13.3

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 3.0.2

**SD card or USB/Octoprint**
SD card

**Describe the bug**
  Filament not loading correctly 
  
**To Reproduce**
Started right after the upgrade tot 3.13.3 so never changed so should be reproducable
  
**Expected behavior**
Filament to extract correctly, I reverted back to 3.13.1 with no adjustments to the printer and no issues occurs. 
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

chrisboer, 2024-03-15:
Sorry did not put in a title and can not change it now.

3d-gussner, 2024-03-15:
@chrisboer Please describe a bit more as  "Filament not loading correctly" can be anything ant any operation.

pawankt90, 2024-03-24:
I have a similar issue with my MK2.5S. The filament sensor doesn't appear to detect filament insertion to load the filament. Unloading seems to work (probably because you can unload via knob control). Reverted my printer to 13.3.1 for now to continue printing.

freswa, 2024-03-31:
Same here. I run a MK3S with sensor v0.4. The sensor values (0,1) read correct in the support section. The voltage goes from 4,7 V down to 1,9 V if I load the filament. The filament sensor section let's me toggle the sensor On/Off, but autoload is N/A as is filament runout.
The sensor test itself fails.
I run into the same issues as the other reporters. autoloading doesn't work. Nothing happens if I put in the filament, but I can confirm the sensor values in the support section flip at the very same time, as does the voltage.

Doesn't happen on 3.13.2 for me.

freswa, 2024-04-04:
Went back to 3.13.1 for the support and up to 3.13.3 again and it's fixed now -.-"

chrisboer, 2024-04-04:
I also went back to v3.13.1 and will not change untill some proper testing is being done on Prusa side !

chrisboer, 2024-04-04:
> @chrisboer Please describe a bit more as "Filament not loading correctly" can be anything at any operation.

If I start printing a new item from SD card it does not load filament even dough filament is correctly inserted in the MMU2S. After downgrading to 3.13.1 without toughing the filament and restarting the same job everything works fine.

freswa, 2024-04-04:
@chrisboer Try to get in touch with support. They have healing effect on your setup, even though you're just re-iterating what you've already done :)

3d-gussner, 2024-04-04:
>I also went back to v3.13.1 and will not change untill some proper testing is being done on Prusa side !
The FW 3.13.3 with MMU3 FW3.0.2 has been tested very firmly and it works for so many users.

Normally when you have the MMU activated in FW 3.13.3 and something is wrong during the MMU operations you should see an Error message with a prusa.io/04nnn link. Similar to this
[image]

> If I start printing a new item from SD card it does not load filament even dough filament is correctly inserted in the MMU2S. After downgrading to 3.13.1 without toughing the filament and restarting the same job everything works fine.

Thanks for the response. So you have a MMU with FW 3.0.2 and the printer with FW3.13.3 and filament is loaded correctly to the MMU.

>does not load filament even dough filament
Can I assume that the MMU does start to load any filament BUT doesn't reach the extruder? Then I would assume that your bowden tube from MMU to Extruder is different. But again you should see an Error message.

Please contact Prusa Live support they can help you.

3d-gussner, 2024-04-04:
> @chrisboer Try to get in touch with support. They have healing effect on your setup, even though you're just re-iterating what you've already done :)

Not sure about the `healing effect` but they are the first you should contact with issues.
The developers are in touch with the Prusa support team and if they can't solve an issue they will contact us or ask to open a github issue.
This github issue tracker is more for feature requests and issues within the firmware.

Prusa-Support, 2024-05-18:
This issue has been silent for some time and the problem doesn't seem to find an explanation on the firmware' side. The lack of consistency and details actually leaves room for any other kind of problem to be solved through a proper troubleshooting session.

Our Customer Support remains available for you to find the most suitable solutions based on individual cases of any kind - where error codes and visual evidence (pictures, videos) will surely come in handy.
https://help.prusa3d.com/article/customer-support_2287

I suggest closing this issue as potentially not related to the firmware and please remember, if you move from FW 3.13.3 to 3.13.2/3.13.1/3.13.0 (and vice versa), you must change the _MMU control board firmware_ accordingly.
https://help.prusa3d.com/article/mmu3-compatibility_470808

Also, in case of problems arose "almost coincidentally" with a firmware installation, legend has it that a factory reset and an additional firmware installation **on both boards** can have a "healing effect" too. 🙂 
https://help.prusa3d.com/article/full-system-refresh-original-prusa-i3_133258
https://help.prusa3d.com/article/factory-reset-mmu_510075

Michele Moramarco
Prusa Research

Prusa-Support, 2024-05-28:
In case of new occurrences of this and similar problems, please get in touch with our Customer Support for troubleshooting.

Michele Moramarco
Prusa Research

closed 2024-05-28 by Prusa-Support