prusa3d/Prusa-Firmware #4638: [BUG] After update to 3.13.3 only English language work

diablo1281, 2024-03-17:
**Printer type** -  MK3S
**Printer firmware version** -  3.13.3

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 3.0.2

**SD card or USB/Octoprint**
  Octoprint (but it doesn't matter)

**Describe the bug**
  Unable to set other language than English. After the update from 3.13.2 to 3.13.3, a message for choosing language appeared. After choosing Polish and accepting it, again it showed a language selection screen (the short before was "Sorting files..."). After choosing English everything goes alright. Trying to change language afterward does the same, and choosing any other is not possible.

Also after the update, the option to enable MMU was reset to OFF... Hope other settings are untouched. Setting everything will be an unnecessary nightmare.
  
**To Reproduce**
  IDK
  
**Expected behavior**
  Should choose and copy the selected language.

sarusani, 2024-03-17:
Did you update the firmware with Octoprint? Octoprint does not support updating the additional languages. 
(Unless you compile your own custom version of avrdude)

You need to update with Prusa Slicer via USB cable to install the languages.

diablo1281, 2024-03-17:
I updated through USB and PrusaSlicer.

sarusani, 2024-03-17:
I just downloaded 3.13.3 and installed it on my MK3S+. I tested polish & german, both loaded without any issues.

Can you flash it again and post the log?

You should see it flashing twice, first the firmware: (flash:w:0)
```
avrdude-slic3r -v -p atmega2560 -c wiring -P COM3 -b 115200 -D -U flash:w:0:C:\Users\Sarusani\Desktop\prusa3d_fw_MK3S_MK3S+_3_13_3_7094.hex:i

avrdude-slic3r: AVR device initialized and ready to accept instructions
Reading | ################################################## | 100% 0.00s

avrdude-slic3r: writing flash (246430 bytes):
Writing | ################################################## | 100% 43.29s

avrdude-slic3r: reading on-chip flash data:
Reading | ################################################## | 100% 31.55s
avrdude-slic3r: verifying ...
avrdude-slic3r: 246430 bytes of flash verified

avrdude-slic3r done.  Thank you.
```

And then the languages: (flash:w:1)

```
avrdude-slic3r -v -p atmega2560 -c arduino -P COM3 -b 115200 -D -u -U flash:w:1:C:\Users\Sarusani\Desktop\prusa3d_fw_MK3S_MK3S+_3_13_3_7094.hex:i

avrdude-slic3r: AVR device initialized and ready to accept instructions
Reading | ################################################## | 100% 0.00s

avrdude-slic3r: writing flash (163958 bytes):
Writing | ################################################## | 100% 19.46s

avrdude-slic3r: reading on-chip flash data:
Reading | ################################################## | 100% 18.40s

avrdude-slic3r done.  Thank you.
```

(I removed a lot of the lines, just check if both end with a successful verify)

diablo1281, 2024-03-19:
Just flashed again and now languages started to work. It's a bit strange, earlier I flashed it 3 or 4 times with multiple restarts of the printer.

I tried to replicate the bug again but without success.

Log:
```
avrdude-slic3r -v -p atmega2560 -c wiring -P COM7 -b 115200 -D -U flash:w:0:C:\Users\diabl\Desktop\prusa3d_fw_MK3S_MK3S+_3_13_3_7094.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Feb 29 2024 at 13:07:16
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : COM7
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
                AVR Part                      : ATmega2560
                Chip Erase delay              : 9000 us
                PAGEL                         : PD7
                BS2                           : PA0
                RESET disposition             : dedicated
                RETRY pulse                   : SCK
                serial program mode           : yes
                parallel program mode         : yes
                Timeout                       : 200
                StabDelay                     : 100
                CmdexeDelay                   : 25
                SyncLoops                     : 32
                ByteDelay                     : 0
                PollIndex                     : 3
                PollValue                     : 0x53
                Memory Detail                 :

                                         Block Poll               Page                       Polled
                  Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                  flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                  lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                  signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

                Programmer Type : Wiring
                Description     : Wiring
                Programmer Model: AVRISP
                Hardware Version: 15
                Firmware Version Master : 2.10
                Vtarget         : 0,0 V
                SCK period      : 0,1 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0,01s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: reading input file "C:\Users\diabl\Desktop\prusa3d_fw_MK3S_MK3S+_3_13_3_7094.hex"
avrdude-slic3r: writing flash (246430 bytes):

Writing | ################################################## | 100% 43,29s

avrdude-slic3r: 246430 bytes of flash written
avrdude-slic3r: verifying flash memory against C:\Users\diabl\Desktop\prusa3d_fw_MK3S_MK3S+_3_13_3_7094.hex:
avrdude-slic3r: load data flash data from input file C:\Users\diabl\Desktop\prusa3d_fw_MK3S_MK3S+_3_13_3_7094.hex:
avrdude-slic3r: input file C:\Users\diabl\Desktop\prusa3d_fw_MK3S_MK3S+_3_13_3_7094.hex contains 246430 bytes
avrdude-slic3r: reading on-chip flash data:

[... 56 lines cut ...]
                SCK period      : 3,3 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0,00s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: reading input file "C:\Users\diabl\Desktop\prusa3d_fw_MK3S_MK3S+_3_13_3_7094.hex"
avrdude-slic3r: writing flash (163958 bytes):

Writing | ################################################## | 100% 19,56s

avrdude-slic3r: 163958 bytes of flash written
avrdude-slic3r: verifying flash memory against C:\Users\diabl\Desktop\prusa3d_fw_MK3S_MK3S+_3_13_3_7094.hex:
avrdude-slic3r: load data flash data from input file C:\Users\diabl\Desktop\prusa3d_fw_MK3S_MK3S+_3_13_3_7094.hex:
avrdude-slic3r: input file C:\Users\diabl\Desktop\prusa3d_fw_MK3S_MK3S+_3_13_3_7094.hex contains 163958 bytes
avrdude-slic3r: reading on-chip flash data:

Reading | ################################################## | 100% 18,39s

avrdude-slic3r: verifying ...
avrdude-slic3r: 163958 bytes of flash verified

avrdude-slic3r done.  Thank you.
```

sarusani, 2024-03-19:
As far as I can see everything looks good on your side. I don't have any idea why it didn't work out before, but glad it works for now.

If you have trouble in the future, please report it again, maybe we can figure it out with some more data.

diablo1281, 2024-03-19:
Sure, I will keep logs if it happens again. Thank you for your help. :)

closed 2024-03-19 by diablo1281

3d-gussner, 2024-03-20:
@diablo1281 Glad it is working. @sarusani Thanks for helping other users, very appreciated. :hugs: