prusa3d/Prusa-Firmware #4640: MMU2S to MMU3 upgrade: Selector cannot home #04115

calipsoii, 2024-03-19:
**Printer type** - MK3.5
**Printer firmware version** -  5.2.1

**MMU upgrade** - MMU2S->MMU3 upgrade
**MMU upgrade firmware version** - 3.0.2+878

**SD card or USB/Octoprint**
USB/Octoprint

**Describe the bug**
After upgrading my MMU2S to an MMU3, the selector cannot home if it is parked >50% of the way to the right. It displays error "Selector cannot home #04115". It WILL home if manually moved <50% of the way to the right. There are at least two other customers who followed the same upgrade path as me and are having the same issue, their comments are at the bottom of this article:
https://help.prusa3d.com/article/selector-cannot-home-04115-mmu_396152
  
**To Reproduce**
With no filament loaded, place the selector (either automatically or manually) 50%+ of the way to the right towards the selector motor. Take any action that triggers a selector home, such as hitting the Restart button on the front of your MK3.5. Your printer will start up successfully, then the MMU3 will attempt to home. The idler will successfully roll on the X axis. The selector will attempt to home, and if it's any closer than 50% of the way towards the selector motor, it will bump 3x against the housing and display error "Selector cannot home #04115".

**Expected behavior**
My expectation is that the selector will be able to home itself no matter where it is parked. This was the behavior on my MMU2S.
  
**Video**

Here is a video of my MK3.5 sitting idle with the selector placed almost all the way to the right. I reboot the machine and the MMU3 fails to home:
https://calipsoii-1d4279c752674f6b5f9b-endpoint.azureedge.net/blobcalipsoii3eb2982b2a/wp-content/uploads/2024/03/mmu3-selector-cannot-home-reboot.mp4

Here is a video of the selector manually moved 66% of the way across its bars towards the selector motor and again failing to home:
https://calipsoii-1d4279c752674f6b5f9b-endpoint.azureedge.net/blobcalipsoii3eb2982b2a/wp-content/uploads/2024/03/mmu3-selector-2thirds-right-1.mp4

Here is a video of the selector moved 33% of the way across it's bars (away from the selector motor) and succeeding:
https://calipsoii-1d4279c752674f6b5f9b-endpoint.azureedge.net/blobcalipsoii3eb2982b2a/wp-content/uploads/2024/03/mmu3-selector-CAN-home-middle.mp4

calipsoii, 2024-03-19:
**Troubleshooting update**: I've managed to seemingly resolve this issue (only been testing for 1 hour now but looks promising) by loosening the 3 machine screws holding the selector motor to the MMU3 body. I loosened each of these 3 machine screws approx. 1/2 a turn. The selector motor now jiggles very slightly when I shake it - it's no longer held securely to the frame. But I can get the selector to home in any location with 100% success.

My theory is that tightening the selector motor to the MMU body positions the long silver leadscrew off-center. Then, as you move to the selector carriage to the right, the leadscrew is being pushed up on the left side until it contacts the MMU body and the friction increases.

I believe this because when the 3 machine screws are fully tightened, the leadscrew is _extremely_ difficult to turn when the selector is close to the motor. But the second you loosen the selector motor screws, the leadscrew becomes quite easy to turn with your fingers, even when the selector is all the way to the right. There seems to be a LOT of friction introduced when the selector motor is tightened to the MMU body, and the stepper motor seems to be having troubles overcoming it and reports that it cannot move.

I don't see any obviously visible difference between the screws being tight:
[image]

And the screws being loose:
[image]

But I can definitely feel with my fingers how much harder the motor leadscrew becomes to turn when fully tightened.

For now I'm going to leave all 3 loose, but this is a poor solution and the entire assembly makes a rattling noise in operation. Can you please attempt to reproduce with an MMU2S->MMU3 upgrade kit and let me know if you see the same behaviour?

3d-gussner, 2024-03-20:
@calipsoii This is the firwware repository for the MK2, MK2.5/S and MK3/S/+. Please reopen your issue here https://github.com/prusa3d/Prusa-Firmware-Buddy as this is for the MINI, MK3.5/9, MK4 and XL running on 32-bit controllers.

calipsoii, 2024-03-20:
> @calipsoii This is the firwware repository for the MK2, MK2.5/S and MK3/S/+. Please reopen your issue here https://github.com/prusa3d/Prusa-Firmware-Buddy as this is for the MINI, MK3.5/9, MK4 and XL running on 32-bit controllers.

Ah, apologies, didn't realize there were two repos going on. I will move this report over there, thank you.

closed 2024-03-20 by calipsoii