prusa3d/Prusa-Firmware #4641: [BUG] Ghost layer shifts (at times) in X axis

SteveKilling, 2024-03-19:
-->
**Printer type** - MK3
**Printer firmware version** -  3.13.3

**MMU upgrade** - no MMU
**MMU upgrade firmware version** 

**SD card used for file transfer

**Describe the bug**
 Two parts were printed of this large duct (5 hr 46 M) successfully. The 3rd part made a shift in the X axis of about 0.6mm at about 75% of the print. This sounds like on of the challenges you were trying to fix in the latest release.
  
**To Reproduce**
  The first two prints were fine. The third had the error.
  
**Expected behavior**
  The first two prints were fine. The third had the error.
  
**G-code**
  I have attached the gcode file as a zip file.

**Video and photo**
The photo shows the two successful prints on the left and the X axis shift on the right.
The video is of the X-shift print.
  
[image]

https://github.com/prusa3d/Prusa-Firmware/assets/164060500/20f14836-52fd-44d3-81ba-eaa5585c9fac

[Fog_Port_0.25mm_PETG_MK3_5h46m.zip](https://github.com/prusa3d/Prusa-Firmware/files/14654358/Fog_Port_0.25mm_PETG_MK3_5h46m.zip)

3d-gussner, 2024-03-21:
@SteveKilling Sorry to hear that, did you contact tech support? Did the printer crashed on X or Y?

Is there a reason you choose the random seams, for me it looks like it generates way more travel moves than needed.
[image]

Can it be that the layer shift happened at the internal overhangs?
While scrolling in Gcode viewer I see that these are printed first and may curl up and cool down before the other lines are printed. 
[image]

Similar object with random seams
[image]
vs aligned seams
[image]
vs nearest seams
[image]

SteveKilling, 2024-03-21:
3d-gussner,
Thanks so much for your reply. I didn't contact tech support since I got the impression in the firmware release that this was the best way to make Prusa contact? No so?

The shift was in the X-axis and is just when the internal overhang shelf turns to vertical.
It looks like it is at this point (image below) - just as the holes are beginning and the shelf is starting the vertical portion.
[image]

Thanks for the suggestion on the seam placement. I have always used random to keep form getting a vertical row of nubs on the model, but hadn't realized the time and motion implications.

If the internal overhanging mini-shelf curled up as you suggest, that might cause a collision that shifts X, but then it recovers and continues? Is that your thought?

Thanks for your help,
Steve

3d-gussner, 2024-03-21:
If you print the same gcode again, please try to get some logfiles by connecting a PC or SBC via USB (before the print) maybe we can see, what is going on.
Please modify your gcode for extended logging with an additional Line `M155 S1 C7` after `G28 W` to report temp, pos and fan.

Mec-mec, 2024-03-21:
I have same problem with Prusa MK3s+ (fw. 3.13.3), there are random layer shifts in stealth mode (seam position aligned).
[image]

3d-gussner, 2024-04-03:
@Mec-mec In Stealth Mode the crash detection is disabled as it isn't reliable. Please verify with the PrusaSlicer Gcode viewer if the print has lot of travel moves.
As you can see above I could find a lot of travel movements and tiny parts printed that may have curled up and caused the shifts.

lordzurp, 2024-04-12:
Hi all

I had the "ghost layer shift" bug with 3.13.2 firmware, so I upgraded to 3.13.3
it was a bit better, but still have 50% of prints shifted. both X and Y crash detections without reason nor physical collision

I tried to revert to 3.13.1 this week, and the few test prints I made seems to be perfect, as usual before upgrading to 3.13.x

**Printer type** - MK3S+, PETG 1.75 filament. Machine recently cleaned and lubricated (new linear bearings)
**Printer firmware version** - 3.13.3

**MMU upgrade** - no MMU
Octoprint 1.9.3, octopi 1.0 and octodash, connected with the UART port directly on the PI 3B+
PrusaSlicer up to date with prusament profiles

worked almost perfectly since 3K hours on meatpack custom firmware before going back on official firmware (3.13.2, for the meatpack support)

Mec-mec, 2024-04-12:
> @Mec-mec In Stealth Mode the crash detection is disabled as it isn't reliable. Please verify with the PrusaSlicer Gcode viewer if the print has lot of travel moves. As you can see above I could find a lot of travel movements and tiny parts printed that may have curled up and caused the shifts.

I rolled back to 3.12.1 firmware, there are no layer shifting.

orbitinstasis, 2024-05-01:
> I rolled back to 3.12.1 firmware, there are no layer shifting.

What features do you miss by doing  this ? 

I'm having too many annoying Y axis phantom crashes.

closed 2024-07-16 by github-actions[bot]