prusa3d/Prusa-Firmware #4643: [ENHANCEMENT]Filament load needs to be 3 state

Fojtik, 2024-03-21:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MK3S, MK3,

**Is your enhancement related to a problem? Please describe.**
  Load filament has 2 options was a filament loaded correctly Yes / No.

It is not sufficient enough. Many times I get into a situation when fillament stuck and both answers are irelevant.
You will need to immediatelly eject.

The situation might be bad cut fillament and stuck, wrong temperature and so on.

The answer Yes is the lie, but it only allows to further fillament eject.

The correct answer No makes a situation even worse.

**Describe the solution you'd like**
There is a neccessity to add 3rd option **eject**.

3d-gussner, 2024-03-21:
@Fojtik Thanks for the idea, we look into that.
I can imagine sometimes the filament load doesn't work correctly and you want to be able to cancel/eject/unload the filament again.

Fojtik, 2024-03-21:
Exactly.
At least I do not want to lie and answer yes that filament is properly loaded.
Answer No means to further move E axis. But if you see that filament is hanged, it is pretty useless or even destructive.

closed 2024-04-11 by 3d-gussner

3d-gussner, 2024-04-12:
@Fojtik This will be in FW 3.14.1 so it will take some time until you see it. Next release will be FW 3.14.0
Kudos to @sarusani to implement it. :clap: