prusa3d/Prusa-Firmware #4648: MK3S+ whith MMU2S smooth sheet

Bizance, 2024-04-02:
**Printer type** - [e.g. MK3S+]
**Printer firmware version** -  [3.13.3 ...]

**MMU upgrade** - [e.g. MMU2S]
**MMU upgrade firmware version** - [e.g. 3.0.2]

**SD card or USB/Octoprint**
  Please let us know if you print via SD card 

**Describe the bug**
  
    we have discovered that with the new software FIRMWARE 3.13.3 (3.0.2) on the MK3S+ +MMU2S with the revo nozzle on the printer it keeps going back to the texture bed even though we select the smooth bed, is anyone else having problems with this

3d-gussner, 2024-04-03:
@Bizance Thanks for reporting this issue, can you please describe the issue step by step
Example:
1. Select Sheet in Settings -> HW Setup -> Steel sheets -> Sheet Smooth1
2. Start SD print (MMU or normal)
3. Check Sheet

and so on

Bizance, 2024-04-04:
We do all these settings to change the sheet, but when we look at print from SD and look at live adjust Z, it says texture sheet and not smooth sheet.

3d-gussner, 2024-06-03:
@Bizance I can't reproduce your issue. When I change the sheets it shows the correct sheet and also values. Can you please take some pictures?

3d-gussner, 2024-07-08:
Closing as no interaction.

closed 2024-07-08 by 3d-gussner