prusa3d/Prusa-Firmware #4653: [BUG] Preheating for (un)load shows 0°C as goal temperature

YolloPlays, 2024-04-04:
**Printer type** - MK3S+ (e3d revo)
**Printer firmware version** - 3.13.3 (for Mk3s+ revo)

**SD card or USB/Octoprint** - Octoprint (1.9.3)

**Describe the bug**
 When trying to unload the filament the goal temperature that the hotend wants to reach is 0°C which is clearly not possible and thus never reached. Therefore the filament can never be unloaded.
When trying to load the filament the goal temp is also 0°C it does not ask for the material. (video included)
 (Work around: Set temp with octo or via settings/preheat)
  
**To Reproduce**
  Unload filament after any print or at any given time after. Happens with filament loaded and without.
  Load filament at any time (with IR sensor and with manually pressing 'Load filament')
  
**Expected behavior**
  I expect it to work like it did in 3.13.2. Preheating to the right temperature.

**Video**
https://github.com/prusa3d/Prusa-Firmware/assets/52836614/10bdc062-6b34-4436-a566-5b2217e8a804

3d-gussner, 2024-04-05:
@YolloPlays Thanks for reporting this issue. Are you sure that with FW 3.13.2 this was working?
We only fixed the Ghost-Layer-shift issue, increased the MMU version number and increased the END_FILE_SECTION see [here](https://github.com/prusa3d/Prusa-Firmware/compare/v3.13.2...v3.13.3) and [here](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.3)

3d-gussner, 2024-04-05:
We have some fixes related to the temperatures for un- and load which will be in FW 3.14.0

YolloPlays, 2024-04-05:
Hi @3d-gussner ,

Those messages seem like they contradict each other. Is the issue on my end or on yours? If it is on my end, is it software or hardware? 

Would you need further information? Therm. Cal. did pass just fine. It prints like it used to.

Yes, I am sure that this didn't happen in .2

Thanks for getting back :)

closed 2024-06-12 by github-actions[bot]