prusa3d/Prusa-Firmware #4654: [BUG]<Enter comprehensive title>

unixcommando, 2024-04-04:
Printer type** - [e.g. MK3S+
Printer firmware version** -  3.13.3

 USB/Octoprint**

My printer has been running flawlessly for about 3 months, this morning I updated Prusaslicer and my printer told me to update the firmware, so I updated to 3.13.3.

I was 3 hours into a 6 hour print when the print stopped and would not continue.  I attempted to send an M600 via the terminal hoping it would initiate a filament change and continue, but it didn't.  After a few minutes of no response, I hit the cancel button on Octoprint and nothing happened.  The nozzle and bed both maintained the set temperature just nothing would happen.  Frustrated I hit the reset button and communication to the printer was restored.

I haven't tried to reproduce, but the print was at Z +2mm.  Not sure it it'll do it again at that spot.

[David_and_Elaine_0.2mm_PETG_MK3S_6h20m.gcode.gz](https://github.com/prusa3d/Prusa-Firmware/files/14878125/David_and_Elaine_0.2mm_PETG_MK3S_6h20m.gcode.gz)

gcode attached.

3d-gussner, 2024-04-05:
@unixcommando Which version of OctoPrint are you using?
Which SBC (RPi) are you using and how is it connected?
Reading your issue it looks to me that OctoPrint lost connection or something else with the host happened as the printer was still heated and doing nothing.
When a host/OctoPrint print is going on the printer only does what it is told to from the host, so it is printing as long the host sends new instructions.
On SD card prints the printer reads line per line instructions.

Which firmware version did you upgrade from?
What did the OcotPrint page looked like? Was it still connected?
Please check OctoPrint SBC if there are any issues reported?

Without any log files this will be hard to troubleshoot. Please enable serial logging in OctoPrint to get a chance to troubleshoot another issue.

unixcommando, 2024-04-05:
Octoprint 1.9.3.
Regards-Bob

3d-gussner, 2024-04-05:
@unixcommando Thanks but you answered one of six questions. What about the rest?

unixcommando, 2024-04-05:
Which version of OctoPrint are you using?

Which SBC (RPi) are you using and how is it connected? Raspberry Pi 4 with 2GB Ram
Reading your issue it looks to me that OctoPrint lost connection or something else with the host happened as the printer was still heated and doing nothing. It was still showing connected in the Octoprint interface, 

When a host/OctoPrint print is going on the printer only does what it is told to from the host, so it is printing as long the host sends new instructions.

On SD card prints the printer reads line per line instructions. 

Which firmware version did you upgrade from? 3.13.1.

What did the OcotPrint page looked like? Was it still connected? The octoprint page looked as expected, the GCODE Viewer showed the code that had been executed up to the point it stopped.

Please check OctoPrint SBC if there are any issues reported? All I see are messages regarding when I unplugged the USB to I could restore firmware 3.13.1
The SBC I've been running about 4 years and had been attached to my Mk2.5s without issue.  When I built the Mk3S+ in Novemeber, rather than dedicate 2 Pis to running Octoprint I am running two instances of Octoprint on the single Pi, this seems to work well, the 2nd instance only seems to take an additional 144M when running.  Yesterday I was only running a single instance for the Mk3S+.  This version of Octoprint I've been running for about 6 weeks without issue and have had both printers running simultaneously without fault.  Yesterday after upgrading the firmware I had a thermal event cancel the print at layer 2 of the gcode I uploaded.  My Ocotprint temp graph showed no temp anomolies, the only change in temps was AFTER the abort.  I didn't report this because I saw others had.  I did see someone had a print stop like mine did, but he reported his printer reset, mine did not, it just sat there doing nothing.
I've reverted to 3.13.1 and was able to complete the print.
Regards-Bob

3d-gussner, 2024-07-08:
@unixcommando Please checkout https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.14.0 and don't forget to also install `Active Prusa HostTimer` OctoPrint Plugin.

Please also run the Thermal Model calibration and PID tuning to ensure that these up to date.

Also useful is the serial log from OctoPrint (frankly it can get quite big during prints).

closed 2024-09-16 by github-actions[bot]