prusa3d/Prusa-Firmware #4660: Problem with Build of downloaded Open Source Prusa firmware.

RogerInHawaii, 2024-04-14:
I'm attempting to download from GitHub the Open Source code for the Prusa firmware, for use on Windows and using Microsoft Visual Studio.

I'm following the section of the README file that's for Windows.

I've got the latest Visual Studio program, have added the CMake Tools plugin,  added Python, and Git Bash (although that seems to get installed as Git CMD.

I'm able to bring the Prusa Firmware up in Visual Studio and have run the

python .\utils\bootstrap.py

to download all dependencies required to build the firmware and I do indeed see a .dependencies folder in the Prusa-Firmware folder.

So far so good.

I'm at the point in t he instruction where it's for Building. And it says . . .

_"To start building a firmware, click the CMake Tools plugin icon on the far left side. You will get a very large list of targets to build. Find the firmware you'd like to build (like MK3S_ENGLISH) and select the small icon which shows "Build" when hovered over."_

Well, there's nothing on the left side of the Visual Studio application window.

On the RIGHT side there is the Solution Explorer with a CMake Targets View.

See the attached image file.

But there's nothing like MK3S_ENGLISH and nothing gets a "Build" option when I hover over it.

So, what do I have to do to actually do a build to create the firmware that can subsequently be loaded into my Prusa MK3?
[image]

gudnimg, 2024-04-14:
The README is using Visual Studio Code, not Visual Studio. The user interface should look like this
[image]

RogerInHawaii, 2024-04-14:
OK, I have never heard of VSCode in my entire career as a software engineer. I've used Visual Studio for decades. Doing a search now I see that VSCode is a basically a "lightweight" version of Visual Studio, which would SEEM to indicate that Visual Studio can also be used in my project, instead of VSCode.

I surely don't want to have to now acquire and learn VSCode, first, in order to proceed with my project.

Is there something comparable in Visual Studio that I can select in order to do the appropriate build of the Prusa Firmware so that it can be then loaded into and run on my Prusa MK3?

gudnimg, 2024-04-14:
I havent used Visual Studio in a while, especially not with CMake. I suspect the configuration files are not compatible with Visual Studio: `https://github.com/prusa3d/Prusa-Firmware/tree/MK3/.vscode` the `.vscode` directory is used by Visual Studio Code and the CMake Tools plugin.

dameningen4914, 2024-04-16:
check this page. you can understand easily.

https://stackoverflow.com/questions/30527522/what-are-the-differences-between-visual-studio-code-and-visual-studio

3d-gussner, 2024-05-02:
Closing due to lack of interaction.

closed 2024-05-02 by 3d-gussner