prusa3d/Prusa-Firmware #4667: Problem with setting everything up with VSCode

RogerInHawaii, 2024-04-17:
I'm having difficulty getting everything set up so that I can use VSCode to edit the Prusa Firmware and then build (make?) it.

I'm following the instructions in the README file  and have installed . . .

_Visual Studio Code
CMake Tools plugin
Python
Git Bash_

and then I clone the repository . . .

git clone https://github.com/prusa3d/Prusa-Firmware

And that seems to go OK. I see a folder for Prusa-Firmware with a lot of folders and files in it.

It then tells me to:

_Open the Prusa-Firmware folder in VScode.

Open a new terminal in VScode (Terminal→New Terminal) and run

python .\utils\bootstrap.py_

which I do. README then says:

_This will download all dependencies required to build the firmware. You should see a .dependencies folder in the Prusa-Firmware folder._

And I do indeed see a .dependencies folder.

README tells me:

_Reload VScode. If all works correctly you should see the VScode automatically configuring the CMake project for you._

I do indeed reload VSCode and my Prusa-Firmware folder is automatically opened .

BUT, it shows a lot of OUTPUT with ERRORS indicated in it. Several **NOENT** plus a **Python3 Not Found**

proc] Executing command: ${workspaceFolder}/.dependencies/cmake-3.22.5/bin/cmake --version
[proc] The command: ${workspaceFolder}/.dependencies/cmake-3.22.5/bin/cmake --version **failed with error**: Error: spawn ${workspaceFolder}/.dependencies/cmake-3.22.5/bin/cmake **ENOENT**
[proc] Executing command: ${workspaceFolder}/.dependencies/cmake-3.22.5/bin/cmake -E capabilities
[proc] The command: ${workspaceFolder}/.dependencies/cmake-3.22.5/bin/cmake -E capabilities **failed with error**: Error: spawn ${workspaceFolder}/.dependencies/cmake-3.22.5/bin/cmake **ENOENT**
[variant] Loaded new set of variants
[kit] Successfully loaded 4 kits from C:\Users\Roger\AppData\Local\CMakeTools\cmake-tools-kits.json
[kit] Successfully loaded 1 kits from c:\Prusa-Firmware\.vscode\cmake-kits.json
[proc] Executing command: c:\Prusa-Firmware/.dependencies/cmake-3.22.5/bin/cmake --version
[proc] Executing command: c:\Prusa-Firmware/.dependencies/cmake-3.22.5/bin/cmake -E capabilities
[proc] Executing command: c:\Prusa-Firmware/.dependencies/cmake-3.22.5/bin/cmake -SC:/Prusa-Firmware -Bc:/Prusa-Firmware/build -G Ninja
[proc] The command: c:\Prusa-Firmware/.dependencies/cmake-3.22.5/bin/cmake -SC:/Prusa-Firmware -Bc:/Prusa-Firmware/build -G Ninja exited with code: 1
[main] Configuring project: Prusa-Firmware 
[proc] Executing command: c:\Prusa-Firmware/.dependencies/cmake-3.22.5/bin/cmake --no-warn-unused-cli -DCMAKE_BUILD_TYPE:STRING=Release -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_TOOLCHAIN_FILE:FILEPATH=c:\Prusa-Firmware/cmake/AvrGcc.cmake -DCMAKE_MAKE_PROGRAM:STRING=c:/Prusa-Firmware/.dependencies/ninja-1.10.2/ninja -SC:/Prusa-Firmware -Bc:/Prusa-Firmware/build -G Ninja
[cmake] Not searching for unused variables given on the command line.
[cmake] -- Could NOT find Python3 (missing: Python3_EXECUTABLE Interpreter) 
[cmake] **CMake Error at cmake/Utilities.cmake:6** (message):
[cmake]   **Python3 not found**.
[cmake] Call Stack (most recent call first):
[cmake]   CMakeLists.txt:2 (include)
[cmake] 
[cmake] 
[cmake] -- Configuring incomplete, errors occurred!
[proc] The command: c:\Prusa-Firmware/.dependencies/cmake-3.22.5/bin/cmake --no-warn-unused-cli -DCMAKE_BUILD_TYPE:STRING=Release -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_TOOLCHAIN_FILE:FILEPATH=c:\Prusa-Firmware/cmake/AvrGcc.cmake -DCMAKE_MAKE_PROGRAM:STRING=c:/Prusa-Firmware/.dependencies/ninja-1.10.2/ninja -SC:/Prusa-Firmware -Bc:/Prusa-Firmware/build -G Ninja **exited with code: 1**

The README file says that if there's problems I can try two things:

1)Type Ctrl+Shift+P and search for CMake: Select a Kit. Select avr-gcc. If none appear, Scan for kits first.

2) If 1) does not work for some reason, as a last resort you can edit the CMake Tools settings. Search for "Additional Kits" and add .vscode/cmake-kits.json to the list.

I tried (1) and I get:

Select a Kit for Prusa-Firmware

[Scan for kits] Search for compilers on this computer
[Unspecified] Unspecified (Let CMake guess what compilers and environment to use)
**avr-gcc Kit for toolchain file c\Prusa-Firmware/cmake/AvrGcc.cmake**
Visual Studio Community 2022 Release - amd64 Using compilers for 17.8.5 (x64 architecture)
Visual Studio Community 2022 Release - amd64_x86 Using compilers for 17.8.5 (x64_x86 architecture)
Visual Studio Community 2022 Release - x86 Using compilers for 17.8.5 (x86 architecture)
Visual Studio Community 2022 Release - x86_amd64 Using compilers for 17.8.5 (x86_x64 architecture)

It does show avr-gcc, but clicking on it does nothing.

I try (2) but when I do a search for "Additional kits" nothing comes up and there's no way (that I can see) to add .vscode/cmake-kits.json to the list
.

So I'm at a loss as to what I need to do how, or what I've done wrong.

I HOPE someone has some suggestions.

RogerInHawaii, 2024-04-17:
Oh, and I should add that if I bring up a CMD window and type:

python --version

it does indeed show the python version (3.12.3).

So it sure looks like python is installed.

3d-gussner, 2024-05-02:
Closing due to lack of interaction.

closed 2024-05-02 by 3d-gussner