prusa3d/Prusa-Firmware #4668: [FEATURE REQUEST] MK3.5 - Display sheet in footer

Xav83130, 2024-04-17:
MK3.5
### **Display sheet in footer**

Would it be possible to add the possibility of displaying the sheet in use in the footer.

In the "_user interface_" --> "_footer_" menu
Add "sheet" selected.

Displays the selected sheet in the footer.

This will be even more useful when you rename the sheets ^^.

JonMrowczynski, 2024-04-18:
Hello Xav83130, I hope that you have been doing well!

It appears that your issue is for the MK3.5. The repository for this can be found [here](https://github.com/prusa3d/Prusa-Firmware-Buddy). As stated in the README for [this repository](https://github.com/prusa3d/Prusa-Firmware?tab=readme-ov-file#prusa-firmware-mk3) "This repository contains the source code and the development versions of the firmware running on the [Original Prusa i3](https://prusa3d.com/) MK3S/MK3/MK2.5S/MK2.5 line of printers."

Maybe the README file for the Buddy firmware repo should be updated to make it clearer that it also contains the firmware for the MK3.5...it doesn't seem very obvious upon initial inspection.

From my understanding of what is usually done in this case, it would be most helpful for you to close this issue here and repost it on the [MK3.5 firmware repository's issue tracker](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues) so that it gets seen by the right people.

3d-gussner, 2024-04-18:
@Xav83130 Closing this issue here as it is in the wrong repository.

closed 2024-04-18 by 3d-gussner